LedgerHQ/speculos #2: Incorrect stack pointer initialization

jleni, 2019-11-19:
First of all, I love speculos!

While using it, noticed that the stack pointer seems to be initialized incorrectly. This is problematic because it is not possible to rely on the stack canary.

A typical `app.map` shows something similar to:

```
               [!provide]                PROVIDE (app_stack_canary, .)
                0x20001ce8                . = (. + 0x4)
 *fill*         0x20001ce4        0x4 00
                0x20001ce8                _stack_validation = .
                0x20002800                . = (_stack_validation + STACK_SIZE)
 *fill*         0x20001ce8      0xb18 00
                0x20001ce8                _stack = (ABSOLUTE (END_STACK) - STACK_SIZE)
                [!provide]                PROVIDE (_stack, (ABSOLUTE (END_STACK) - STACK_SIZE))
                0x20002800                _estack = ABSOLUTE (END_STACK)
                [!provide]                PROVIDE (_estack, ABSOLUTE (END_STACK))
```

so `sp` is expected to be in that range.

As soon as launching gdb, `sp` is out ot range:
```
Breakpoint 1, main () at src/main.c:21
-exec info stack
#0  main () at src/main.c:21
#1  0x00010ff8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

$sp
0x20003000
```

@yhql pointed me to this line
https://github.com/LedgerHQ/speculos/blob/b529cd09013bf7dce11bd7db80196d650a26c650/src/launcher.c#L504

Digging a bit into the code, I noticed the following:
https://github.com/LedgerHQ/speculos/blob/b529cd09013bf7dce11bd7db80196d650a26c650/src/launcher.c#L30-L32

As a workaround, I am patching `DATA_VSIZE` with `0x1000`, if you think it makes sense I can make a quick PR. So at least, the canary is triggered properly. But I guess in the long term, I guess it would be good to use `_estack` for this.

greenknot, 2019-11-20:
Thanks for the bug report!

I just created the PR #3 to retrieve the stack address from the ELF binary, does it fix the issue?

jleni, 2019-11-20:
cool! Thanks!
I will test in the next couple of hours and get back to you

jleni, 2019-11-20:
Hi, I tried PR #3 but I am not sure it is correct.
Now things start at 0x20001800 but I was actually expecting 0x20002800
Maybe I am missing something?

For instance:
```
Non-debugging symbols:
0x20001800  _bss
0x20001ce4  _ebss
0x20001ce4  app_stack_canary
0x20001ce8  _stack_validation
0xd0000000  _data
0xd0000000  _edata
0x20001800  _bss
0x20001ce4  _ebss
0x20001ce4  app_stack_canary
0x20001ce8  _stack_validation
0xd0000000  _data
0xd0000000  _edata
```
so canary is at `0x20001ce4` and when starting the app, `$sp` starts at `0x20001800`

jleni, 2019-11-20:
It seems to working well after the last changes! 
Thanks for the great support!

closed 2019-11-20 by greenknot