LedgerHQ/speculos #5: Any call to the SDK hangs if gdb is attached

eehlers, 2019-12-16:
Any call to the SDK hangs if gdb is attached.  I am able to debug the emulator fine prior to calling in to the SDK.

Here are the steps that I have followed:

- For nanos-secure-sdk I have checked out branch webusb-fix (1.5?)

- For ledger-app-btc I use repo https://github.com/LedgerHQ/ledger-app-btc.git, branch master.  I make the following changes to the makefile:
```
#DEBUG = 0
DEBUG = 1

#CFLAGS   += -O3 -Os
CFLAGS   += -g

#LDFLAGS  += -O3 -Os
LDFLAGS  += -g
```
- I put PRINTF() statements in functions btchip_apdu_hash_input_start() and btchip_apdu_hash_input_finalize_full().

- First, as a sanity check, I run the emulator with no debugging:

`./speculos.py --sdk 1.5 apps/btc.ledger.master.elf`

- From btchip-python I send a TX to the emulator.  Within the emulator I sign the TX.  In the stdout I see the PRINTF statements from btchip_apdu_hash_input_start() and btchip_apdu_hash_input_finalize_full(), so I know that these functions are hit.

- Now I run the emulator in the debugger:
```
./speculos.py --sdk 1.5 -d apps/btc.ledger.master.elf &
./tools/debug.sh apps/btc.ledger.master.elf

gdb$ b io_exchange
Breakpoint 2 at 0x40005f50: file /home/erik/projects/ledger/nanos-secure-sdk/src/os_io_seproxyhal.c, line 957.
gdb$ b btchip_apdu_hash_input_start
Breakpoint 3 at 0x400011bc: file src/btchip_apdu_hash_input_start.c, line 32.
gdb$ b btchip_apdu_hash_input_finalize_full
Breakpoint 4 at 0x40000d34: file src/btchip_apdu_hash_input_finalize_full.c, line 498.
gdb$ c
Continuing.
```
Then from btchip-python I send a TX to the emulator.  gdb breaks in io_exchange(), I hit c(ontinue), after that gdb never returns.  The emulator remains responsive to user button presses but does not receive the TX that has been sent to it.  Through further analysis I see that the process hangs in file os_io_seproxyhal.c, in function io_exchange(), near line 1130: `for (;;) { ... }`  It calls in to the SDK and never returns.  My PRINTF() statements in functions btchip_apdu_hash_input_start and btchip_apdu_hash_input_finalize_full are never executed.

eehlers, 2019-12-16:
On slack, Juan Leni suggested `set bracktrace limit 50`, I tried that, no change.

greenknot, 2019-12-17:
Thanks for the detailed bug report, I'll give a look at it.

greenknot, 2019-12-17:
FTR (I don't know if it can help), you can append the `--trace` parameter to display each syscall made: `./speculos.py --trace --sdk 1.5 -d apps/btc.ledger.master.elf`.

greenknot, 2019-12-17:
I can't reproduce the issue. Could you please share `btc.ledger.master.elf` and the btchip-python command?

eehlers, 2019-12-17:
Many thanks for getting back to me.

Here is the trace output from two iterations:

1) Without debugging:

https://pastebin.com/Nf3FSfUY

- I start the emulator:

`./speculos.py --trace --sdk 1.5 apps/btc.ledger.master.elf`

- From btchip-python I send a TX to the emulator.

- Within the emulator I hit right arrow.  It signs the TX and sends it back to my python script, which exits successfully.

- Within the emulator I hit Q.

2) With debugging:

https://pastebin.com/kH0TPCaX

- I start the emulator:
```
erik@laptop3:~/projects/ledger/speculos$ ./speculos.py --trace --sdk 1.5 -d apps/btc.ledger.master.elf &
[1] 19814
erik@laptop3:~/projects/ledger/speculos$ QPixmap::scaled: Pixmap is a null pixmap

erik@laptop3:~/projects/ledger/speculos$ ./tools/debug.sh apps/btc.ledger.master.elf
The target architecture is assumed to be arm
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
--------------------------------------------------------------------------[regs]
  R0:  0x00000000  R1: 0xFFFEF2AE  R2:  0xFFFEF2C3  R3:  0x00000000
  R4:  0x00000000  R5: 0x00000000  R6:  0x00000000  R7:  0x00000000
  R8:  0x00000000  R9: 0x00000000  R10: 0x00201004  R11: 0x00000000 
  R12: 0x00000000  SP: 0xFFFEF100  LR:  0x00000000  PC:  0x000101E8  n z c v q j e a i f T 
--------------------------------------------------------------------------[code]
   0x101e9:	mov.w	r11, #0
   0x101ed:	mov.w	lr, #0
   0x101f1:	pop	{r1}
   0x101f3:	mov	r2, sp
   0x101f5:	push	{r2}
   0x101f7:	push	{r0}
   0x101f9:	ldr.w	r12, [pc, #16]	; 0x1020c
   0x101fd:	str.w	r12, [sp, #-4]!
--------------------------------------------------------------------------------
0x000101e8 in ?? ()
add symbol table from file "apps/btc.ledger.master.elf" at
	.text_addr = 0x40000000
Breakpoint 1 at 0x40000000: file src/main.c, line 2632.
[*] using SDK version 1.5
[*] patching svc instruction at 0x40006f60
--------------------------------------------------------------------------[regs]
  R0:  0x00000000  R1: 0x00000000  R2:  0x20001800  R3:  0x20002800
  R4:  0x40000001  R5: 0x00000000  R6:  0xFFFEEFD8  R7:  0xFFFEEF70
  R8:  0x0021600C  R9: 0x20001800  R10: 0x00201004  R11: 0x00000000 
  R12: 0x000000DF  SP: 0x20002800  LR:  0x00011051  PC:  0x40000000  n z C v q j e a i f T 
--------------------------------------------------------------------------[code]
   0x40000001 <main>:	push	{r4, r5, r6, r7, lr}
   0x40000003 <main+2>:	sub	sp, #44	; 0x2c
   0x40000005 <main+4>:	cpsie	i
   0x40000007 <main+6>:	cmp	r0, #0
   0x40000009 <main+8>:	beq.n	0x40000020 <main+32>
   0x4000000b <main+10>:	mov	r4, r0
   0x4000000d <main+12>:	movs	r0, #1
   0x4000000f <main+14>:	lsls	r0, r0, #8
--------------------------------------------------------------------------------

Breakpoint 1, main (arg0=0x0) at src/main.c:2632
2632	__attribute__((section(".boot"))) int main(int arg0) {
gdb$ b btchip_apdu_hash_input_start
Breakpoint 2 at 0x400011bc: file src/btchip_apdu_hash_input_start.c, line 32.
gdb$ b btchip_apdu_hash_input_finalize_full
Breakpoint 3 at 0x40000d34: file src/btchip_apdu_hash_input_finalize_full.c, line 498.
gdb$ c
Continuing.
```
- From btchip-python I send a TX to the emulator.

gdb never returns.  The emulator remains responsive to user input, but never receives the TX that has been sent to it.  The stdout indicates that the python script has gone into an endless loop.

- Within the emulator I hit Q.  The emulator exits and the Python script crashes.

eehlers, 2019-12-17:
> Could you please share ... the btchip-python command?

I am calling btchip-python from a script which has some dependencies.  I will try to reduce it down to a minimal example, this might help to isolate the problem.  I will get back to you as quickly as I can.

greenknot, 2019-12-17:
No worries, just share the `btc.ledger.master.elf` binary please :)

eehlers, 2019-12-17:
- Here is the link to the binary:

https://sites.google.com/site/general7533/home/files/btc.ledger.master.elf

- Here is my build environment for ledger-app-btc.  As mentioned before, I changed a couple of flags in the makefile, and added a couple of PRINTF statements.  I build with `COIN=bitcoin make`.

```
erik@laptop3:~/projects/ledger/ledger-app-btc.ledger$ git status
On branch master
Your branch is up-to-date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   Makefile
	modified:   src/btchip_apdu_hash_input_finalize_full.c
	modified:   src/btchip_apdu_hash_input_start.c

no changes added to commit (use "git add" and/or "git commit -a")

erik@laptop3:~/projects/ledger/ledger-app-btc.ledger$ git diff
diff --git a/Makefile b/Makefile
index d855a87..eb2f3ff 100755
--- a/Makefile
+++ b/Makefile
@@ -231,7 +231,8 @@ DEFINES                 += HAVE_UX_FLOW
 endif
 
 # Enabling debug PRINTF
-DEBUG = 0
+#DEBUG = 0
+DEBUG = 1
 ifneq ($(DEBUG),0)
 
         ifeq ($(TARGET_NAME),TARGET_NANOX)
@@ -265,12 +266,14 @@ endif
 CC       := $(CLANGPATH)clang
 
 #CFLAGS   += -O0
-CFLAGS   += -O3 -Os
+#CFLAGS   += -O3 -Os
+CFLAGS   += -g
 
 AS     := $(GCCPATH)arm-none-eabi-gcc
 
 LD       := $(GCCPATH)arm-none-eabi-gcc
-LDFLAGS  += -O3 -Os
+#LDFLAGS  += -O3 -Os
+LDFLAGS  += -g
 LDLIBS   += -lm -lgcc -lc
 
 # import rules to compile glyphs(/pone)
diff --git a/src/btchip_apdu_hash_input_finalize_full.c b/src/btchip_apdu_hash_input_finalize_full.c
index 34bc53e..706074d 100644
--- a/src/btchip_apdu_hash_input_finalize_full.c
+++ b/src/btchip_apdu_hash_input_finalize_full.c
@@ -495,6 +495,7 @@ unsigned short btchip_apdu_hash_input_finalize_full_internal(
 }
 
 unsigned short btchip_apdu_hash_input_finalize_full() {
+PRINTF("btchip_apdu_hash_input_finalize_full\n");
     PRINTF("state=%d\n", btchip_context_D.outputParsingState);
     unsigned short sw = btchip_apdu_hash_input_finalize_full_internal(
         &btchip_context_D.transactionSummary);
diff --git a/src/btchip_apdu_hash_input_start.c b/src/btchip_apdu_hash_input_start.c
index caf4704..82cf80e 100644
--- a/src/btchip_apdu_hash_input_start.c
+++ b/src/btchip_apdu_hash_input_start.c
@@ -29,6 +29,7 @@
 
 unsigned short btchip_apdu_hash_input_start() {
     unsigned char apduLength;
+PRINTF("btchip_apdu_hash_input_start\n");
     apduLength = G_io_apdu_buffer[ISO_OFFSET_LC];
 
     SB_CHECK(N_btchip.bkp.config.operationMode);
erik@laptop3:~/projects/ledger/ledger-app-btc.ledger$ 
```

- Below is the python script that I use to test the emulator.  I run the script like so:
```
virtualenv venv
source venv/bin/activate
pip install ecdsa

PYTHONPATH=/home/erik/projects/ledger/btchip-python LEDGER_PROXY_ADDRESS=127.0.0.1 LEDGER_PROXY_PORT=9999 python3 s.py
```

```
(venv) erik@laptop3:~/projects/ledger/test/test03$ cat s.py 
#!/usr/bin/env python3

from btchip import btchip
from btchip.btchipComm import getDongle

dongle = getDongle(True)
app = btchip.btchip(dongle)

segwit_inputs = [{'sequence': 'ffffffff',
        'value': b'\xf7\xcdbN\x8d\xd9\xb1\xe6\x03\x88\xd9E\x97\x10\xf7}z\x01\xb7\xbc'
                b'\x92\xae\xe7\xabFT\xe9\xfe\xbd\xd5\x1ac\x01\x00\x00\x00'
                b'\x00\xe9\xa45\x00\x00\x00\x00',
        'witness': True}]

app.startUntrustedTransaction(True, 0, segwit_inputs, bytearray(b''), 2)

tx_bytes = b'\x02\x00\x00\x00\x01\xf7\xcdbN\x8d\xd9\xb1\xe6\x03\x88\xd9E\x97\x10\xf7}z\x01\xb7\xbc\x92\xae\xe7\xabFT\xe9\xfe\xbd\xd5\x1ac\x01\x00\x00\x00\x00\xff\xff\xff\xff\x01\x00\x08\xaf/\x00\x00\x00\x00\x16\x00\x14\x00 \xcb\xda6s\x11\x8d\x0c\x96<\xc5\xcft\x9ad\xc6\xf9\xb4\x99\x00\x00\x00\x00'

app.finalizeInput(b"DUMMY", -1, -1, '', tx_bytes)
```

If I run the emulator without the debugger, the script completes successfully.  If I run the emulator with the debugger, the script hangs as described previously.

eehlers, 2019-12-17:
I am able to recreate the problem also with the binary `apps/btc.elf` that is included in the repo.

I tested this with both python 2 and 3, no difference.

I am on Ubuntu 18.04.3.  I am using gdb-multiarch version 8.1.0.20180409-git.

eehlers, 2019-12-17:
I tried it with gdb 8.3, no change.

eehlers, 2019-12-17:
Tried gdb 7.12.1, no change.

greenknot, 2019-12-17:
Thanks! I can indeed reproduce the issue.

The following patch seems to fix it:
```diff
diff --git a/mcu/seproxyhal.py b/mcu/seproxyhal.py
index 6161b2e..d0707c4 100644
--- a/mcu/seproxyhal.py
+++ b/mcu/seproxyhal.py
@@ -110,7 +110,7 @@ class SeProxyHal:
         if tag == SEPROXYHAL_TAG_GENERAL_STATUS:
             if int.from_bytes(data[:2], 'big') == SEPROXYHAL_TAG_GENERAL_STATUS_LAST_COMMAND:
                 screen.screen_update()
-
+            return
             if len(self.queue) > 0:
                 # if we've received events from the outside world, send them
                 # whenever the SE is ready (i.e. now as we've received a
@@ -153,8 +153,8 @@ class SeProxyHal:
             sys.exit(0)
 
     def _queue_event_packet(self, tag, data=b''):
-        # self._send_packet(tag, data)
-        self.queue.append((tag, data))
+        self._send_packet(tag, data)
+        #self.queue.append((tag, data))
 
     def handle_button(self, button, pressed):
         '''Forward button press/release from the GUI to the app.'''
```
Unfortunately, this patch breaks the `io_heartbeat` syscall. If you can confirm that this workaround also prevents the hang on your machine, I'll try to develop a correct fix.

eehlers, 2019-12-17:
That patch does appear to fix the problem.  Although the process now runs instanely slow, like you can see the text print to stdout one character at a time.  Big thank you for your response, I will wait for an update.

greenknot, 2019-12-20:
Can you test the PR #7 and confirm that it fixes the issue?

eehlers, 2019-12-20:
I tested that PR, I am still having trouble, I will provide you with a test case.

eehlers, 2019-12-30:
The debugger seems now to be working correctly for binaries built from the master branch and running on mainnet.  In a feature branch, Nicolas customized the `btchip_apdu_hash_input_finalize_full()` function for us (on testnet) and I am still unable to debug that binary.  However I have no easy way to recreate that problem for you, and it may be a separate issue.

I would suggest that you mark the current problem as resolved, and later I will try to find a simple way to reproduce for you the other problem that we are having.

Thank you very much for all of your help, it is much appreciated.

eehlers, 2019-12-30:
P.S. When I say "master branch", what I mean is that the debugger from repo `speculos`, branch `gdb-hang`, is able to debug binaries built on repo `ledger-app-btc`, branch `master`.  It fails to debug a binary built on Nicolas's repo in a feature branch, but that might be a different problem.

PR #7 (2019-12-31): seproxyhal: wait for a status before sending an event
It should fix #5
commit be45a25e (2019-12-31): seproxyhal: wait for a status before sending an event

It should fix #5
closed 2019-12-31 by greenknot