LedgerHQ/speculos #6: Text UI broken on master - "failed to emulate syscall 0x6000721c"

eehlers, 2019-12-20:
If I run the binary from the repo with `--text`, it works as I expect:

    ./speculos.py --text apps/btc.elf

If I do the same with a binary that I built myself from the master branch, it fails with the following error:

    launcher: failed to emulate syscall 0x6000721c

greenknot, 2019-12-20:
`SYSCALL_io_seproxyhal_spi_send_ID_IN` (`0x6000721c`) isn't defined for version 1.6 of the SDK (which is the default). What SDK version did you use to build the app? 

I guess you either have to build your app against SDK 1.6, or use the `--sdk 1.6` parameter on the command line.

eehlers, 2019-12-20:
I observe the opposite of what you describe.

If I run the binary from the repo with `--text`, it works as I expect:

    ./speculos.py --text apps/btc.elf

I have not specified the SDK version.  It defaults to 1.6?

When I build the binary myself from the master branch, I check out branch `webusb-fix` of the SDK.  This is version 1.5?  I run it like so:

    ./speculos.py --sdk 1.5 --text apps/btc.ledger.master.elf

This fails as described before.  So it seems to me that it works for 1.5 and not for 1.6.

greenknot, 2019-12-20:
The default SDK version is indeed 1.6: https://github.com/LedgerHQ/speculos/blob/f36724624e783e5f4cebac205b326536f4187270/speculos.py#L110

I can reproduce the issue with the following command line (app built against SDK 1.5 and ran with version 1.6):
```
$ ./speculos.py --sdk 1.6 apps/nanos#btc#1.5#00000000.elf
[*] using SDK version 1.6
[*] patching svc instruction at 0x400065d8
launcher: failed to emulate syscall 0x6000721c
```

However, it works with:
```
./speculos.py --sdk 1.5 apps/nanos#btc#1.5#00000000.elf
./speculos.py --sdk 1.5 --text apps/nanos#btc#1.5#00000000.elf
```

The error `failed to emulate syscall 0x6000721c` can't happen with `--sdk 1.5`.

eehlers, 2019-12-20:
Thanks for getting back to me.  My mistake, you are right.  I only get `failed to emulate syscall 0x6000721c` if I use the default version of the SDK (1.6):

    ./speculos.py --text apps/btc.ledger.master.elf

If I specify SDK version 1.5...

   ./speculos.py --sdk 1.5 --text apps/btc.ledger.master.elf

...then it fails with a different error:

    [*] using SDK version 1.5
                         [*] patching svc instruction at 0x40006f60
                                                                   CTraceback (most recent call last):
                        File "./speculos.py", line 171, in <module>
                                                                       screen.display(apdu, seph,
    button_tcp=button_tcp, color=args.color, model=args.model, ontop=args.ontop, rendering=render_method, vnc=vnc,
    keymap=args.keymap, pixel_size=zoom)
       File "/home/erik/projects/ledger/speculos/mcu/screen_text.py", line 169, in display
              display.run()
                             File "/home/erik/projects/ledger/speculos/mcu/screen_text.py", line 161, in run
                                self.notifiers[fd].can_read(fd, self)
                                                                       File "/home/erik/projects/ledger/speculos/mcu
    /seproxyhal.py", line 155, in can_read
                                                                              if screen.rendering ==
    RENDER_METHOD.PROGRESSIVE:
                                               AttributeError: 'TextScreen' object has no attribute 'rendering'
                               launcher: read from seph fd failed: Connection reset by peer

Is that expected?  Below is the link to the binary that I am using.  I built it using branch `master` in repo `ledger-app-btc`, branch `webusb-fix` in repo `nanos-secure-sdk`.

    https://sites.google.com/site/general7533/home/files/btc.ledger.master.elf

yhql, 2019-12-20:
Hello, I could not reproduce this bug using the `webusb-fix` branch of the SDK, and compiling the lastest bitcoin app. So it's a bit of a shot in the dark, but could you try with this branch : https://github.com/LedgerHQ/speculos/tree/WIP-fix-%236 ?

eehlers, 2019-12-31:
I just rebuilt and reran everything from scratch and I am no longer getting the error (repo `ledger-app-btc`/ branch `master`, repo `speculos.git`/ branch `master`).  I can't figure out what would have changed on my side to fix the error.

However it does seem that there is a bug in repo `speculos.git`/ branch `master`.  In script `screen_text.py`, class `TextScreen` passes `self` to function `can_read()`, which inspects attribute `rendering`, which is undefined.

The fix that you applied in branch `WIP-fix-#6` appears to me to be a sensible solution.  If it's all right with you, I propose that you merge the branch and mark this resolved.

Mnay thanks for your help.

commit bce4da42 (2019-12-31): mcu: add missing rendering field to the text screen

Fix #6
closed 2019-12-31 by greenknot