LedgerHQ/speculos #10: launcher: failed to emulate syscall xxxxxx

jleni, 2020-01-04:
At the moment, any missing syscall emulation results in the emulator terminating.

It would be nice that missing syscall would trigger a default handler that returns an error to the app without crashing the emulator.

greenknot, 2020-01-06:
The value returned by syscalls is a constant (`SYSCALL_name_ID_OUT`) which is actually never used. One can return a different value (just set `retid` [here](https://github.com/LedgerHQ/speculos/blob/a547c7f4000d6334d9276e82db6eabecb5fa72f1/src/emulate.c#L280)), unfortunately there is no way for the app check that value since [`SVC_Call` returns nothing](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/syscalls.c#L9).

I guess an exception might be thrown though?

greenknot, 2020-01-06:
Even if it doesn't solve the issue, we should develop a script to iterate through every syscall of each SDK version to tell which syscalls are missing.

jleni, 2020-01-06:
yes, I was thinking of a default handler that throws an exception. This is not urgent anyway.

closed 2022-07-05 by greenknot