LedgerHQ/speculos #11: Concatenation of APDU messages is not supported

patriksletmo, 2020-01-07:
Longer APDU messages are typically read into `G_io_apdu_buffer` by concatenating multiple smaller messages that are received in `G_io_seproxyhal_spi_buffer`. It seems like this functionality relies on `io_seproxyhal_spi_is_status_sent` to return a non-zero value when more data is available (i.e. sent on SPI but not yet read into `G_io_seproxyhal_spi_buffer`). I think that [this code](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/os_io_seproxyhal.c#L1186-L1194) may be related.

The current implementation of this function always returns zero, which effectively disables support for message concatenation (see [/src/emu_seproxyhal.c](https://github.com/LedgerHQ/speculos/blob/be45a25e5af7c52b733d9deb781ca3bc656cf600/src/emu_seproxyhal.c#L57)).

With message concatenation disabled, received messages risk to be cut off if `G_io_apdu_buffer` is larger than `G_io_seproxyhal_spi_buffer`. In these cases, the cut off part is filled with zeroes.

The problem can be mitigated on the app side by defining `IO_SEPROXYHAL_BUFFER_SIZE_B` to be sufficiently large. However, the current implementation creates confusion as to why apps behave the way they do when tested using Speculos.

greenknot, 2020-01-16:
Thanks for the bug report!

I tried to understand how `io_seproxyhal_spi_is_status_sent` is supposed to work and created PR #15 . Could you please tell me if it fixes the issue (I'm really not sure about it)?

If the fix isn't correct, can you share either a code snippet which reproduces the issue (or event better app.elf itself) along a client script?

patriksletmo, 2020-01-17:
Thank you for taking your time to look at this issue!

Since reporting the issue I'm not certain that the implementation of `io_seproxyhal_spi_is_status_sent` is the cause of the inconsistent behavior I have been observing. I have tried testing with your patch but it doesn't seem to address this problem.

To make reproducing easier I have created a small example app that shows the difference in behavior using the emulator in contrast to a real Nano S device. You can find the source code together with a pre-built binary and a client script [here](https://github.com/patriksletmo/speculos-reproduce-11). Start the app in the emulator and run `./test.sh` to test the behavior. You can find expected output and further details in the repository README.

greenknot, 2020-01-24:
Thanks a lot for the example app, it helped a lot.

On a real device, [io_usb_hid_receive()](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/os_io_usb.c#L60) is called to concatenate USB chunks into `G_io_apdu_buffer`.

In speculos, packets are [sent as CAPDUs](https://github.com/LedgerHQ/speculos/blob/c379f91240ac7747e345dd700c3e5c850ff4c553/mcu/seproxyhal.py#L201). The app calls `io_exchange()` and the packet is received thanks to a call to [io_seproxyhal_spi_recv()](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/os_io_seproxyhal.c#L1298) (which is emulated by speculos to receive the packet through TCP). The maximum packet length is thus `sizeof(G_io_seproxyhal_spi_buffer)`. [io_seproxyhal_handle_capdu_event()](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/os_io_seproxyhal.c#L297) is eventually called to [fill G_io_apdu_buffer](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/os_io_seproxyhal.c#L254).

I don't know how to solve that issue (and my analysis might be wrong) without patching the SDK.

greenknot, 2020-01-24:
Actually, even if more difficult one can send `SEPROXYHAL_TAG_USB_EP_XFER_EVENT` instead of `SEPROXYHAL_TAG_CAPDU_EVENT` (from [mcu/seproxyhal.py](https://github.com/LedgerHQ/speculos/blob/c379f91240ac7747e345dd700c3e5c850ff4c553/mcu/seproxyhal.py#L201)). It requires some work but should be doable.

greenknot, 2020-02-06:
@patriksletmo I hope the PR #21 fixes this issue for good. Could you give it a try?

patriksletmo, 2020-02-07:
@greenknot Thanks! I tested the new code and it appears that it is a large step in resolving this issue. However, while verifying the functionality in a real app I noticed that my messages were rejected. After some investigation it appears that the changes in #21 introduce corruption to messages with a size of at least 90 bytes.

I have [updated the demo app](https://github.com/patriksletmo/speculos-reproduce-11/commit/99da1076aff5cda5bb33b410072f6f70d7c2584b) with functionality to reproduce this issue.

greenknot, 2020-02-07:
Indeed, chunk sizes were invalid (4b2327fbdeea1f16c2356386ba559c0bd0d17cd9). Thanks for your help!

closed 2020-02-07 by greenknot

patriksletmo, 2020-02-07:
Seems to work much better now! Thank you for resolving this!

PR #15 (2020-02-03): seproxyhal: implement sys_io_seph_is_status_sent
Fix #11.
commit a057f6c2 (2020-02-06): forward packets using USB messages instead of CAPDUs

Fix #11.