LedgerHQ/speculos #17: Add PRINTF analog that works regardless of IO status

teams2ua, 2020-01-20:
PRINTF only works when USB is ready and it is SE turn to send messages. It may work for big number of cases, but will not work for some test applications when USB is not needed, or during experimenting with IO system itself.
Additional syscall could be added like `SPECULOS_LOG(...)` that will work only in Speculos, and output into console regardless the current state of IO.

greenknot, 2020-01-21:
Thanks for the bug report! FTR, this behaviour was introduced in the version  1.6.0 of the SDK. It is due to the following lines in `mcu_usb_prints()`:
```C
if(USBD_Device.dev_state != USBD_STATE_CONFIGURED){
  return;
}
```
I'll try to find a workaround.

greenknot, 2020-02-07:
Side effect of PR #21: it also fixes this issue.

closed 2020-02-07 by greenknot