LedgerHQ/speculos #33: Unable to run VNC server with --display=qt inside Docker

DCRichards, 2020-03-06:
## Issue
Running the emulator from `docker-compose` with `--display qt` fails with numerous errors. 

## Replication
Run the following `--model nanos ./apps/btc.elf --sdk 1.6 --seed secret --display qt --apdu-port 40000 --vnc-port 41000 --button-port 42000"` as the command. At first there will be a number of libs missing:

* Python: `pyqt5`
* Apt: `libgl1-mesa-glx`, `libglib2.0-0`, ` libdbus-1-3` and `libfontconfig`

After these are installed, the error output is as follows:

```bash
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl, xcb.

[*] using SDK version 1.6
vnc_server: readall: connection closed
```

Running with `export QT_DEBUG_PLUGINS=1` yields only a little more info:

```bash
Got keys from plugin meta data ("xcb")
QFactoryLoader::QFactoryLoader() checking directory path "/usr/local/bin/platforms" ...
loaded library "/usr/local/lib/python3.8/site-packages/PyQt5/Qt/plugins/platforms/libqxcb.so"
qt.qpa.xcb: could not connect to display qt
```

greenknot, 2020-03-06:
Running a GUI application inside a Docker container requires to expose the X11 socket to the container (with `docker run -v /tmp/.X11-unix:/tmp/.X11-unix` for instance).

It might be a bit easier to use a VNC client (eg. remmina) from the host and connect to the VNC port (here `41000`).

DCRichards, 2020-03-06:
Ah I see, the vnc server runs in all cases, regardless of `--display`. It looks like it's running and I can attempt to talk to `vnc://localhost:41000` but it just hangs on connection.

[image]

greenknot, 2020-03-06:
A coworker of mine told me that the native MacOS VNC client doesn't seem to work with our implementation (we need to investigate the issue), but VNC Viewer works well.

DCRichards, 2020-03-06:
Awesome, thank you @greenknot, can confirm it worked with [VNC Viewer](https://www.realvnc.com/en/connect/download/viewer/macos/).

[image]

Would you like me to raise an issue for the OSX Screen Sharing issue or PR an update to the docs to mention the issue?

greenknot, 2020-03-06:
Please open a new issue regarding the default MacOS VNC client, I'm closing this one. Thanks!

closed 2020-03-06 by greenknot