LedgerHQ/speculos #34: Unable to connect to VNC server using MacOS Screen Sharing application

DCRichards, 2020-03-06:
## Issue
Connecting to the VNC server from MacOS Screen Sharing app hangs and does not connect.

[image]

## Replication

1. Start the server by running `docker-compose up` with the default options
2. Open **Screen Sharing** and attempt to connect to `vnc://localhost:41000`
3. Asks for a password, hit enter with no password
4. Hangs with no output

## System Info

* MacOS 10.14.6
* Docker 2.2.0.3

## Issue links

For context, this was initially discussed in #33.

greenknot, 2020-03-12:
Workaround: use a VNC password through `--vnc-password` (#36).

The VNC server has to be re-built (`make -C build/vnc/`).

closed 2020-03-12 by greenknot

DCRichards, 2020-03-12:
Awesome, thanks for working on this!