LedgerHQ/speculos #38: Speculos constantly hangs when run in travis

achow101, 2020-03-18:
In [HWI](https://github.com/bitcoin-core/HWI), we use speculos to test that the Ledger Nano S implementation works. However when these tests are run in Travis CI, speculos routinely hangs resulting in a test timeout. There is no particular place that this happens, it just seems to always happen at some point during the test suite.

greenknot, 2020-03-19:
Hello Andrew,

Thanks for the bug report. I assume that Ledger Nano tests are in this [script](https://github.com/bitcoin-core/HWI/blob/master/test/test_ledger.py), I'll try to reproduce the issue. Meanwhile, do you have logs I can look at (I failed at finding any of them on [Travis website](https://travis-ci.org/github/bitcoin-core/HWI/))?

greenknot, 2020-03-19:
By the way, your speculos version seems a bit outdated (the arguments were modified a few months ago in d8df3ee4287b2529e98b395e28e5008d0fd6ff97). Does the issue still exist with the last version?

achow101, 2020-03-19:
Yes, speculos was outdated. It is being updated in https://github.com/bitcoin-core/HWI/pull/311 and the issue still exists. We don't normally save the speculos logs, but one of the runs for https://github.com/bitcoin-core/HWI/pull/311 was outputting the speculos log for debugging. For example: https://travis-ci.org/github/bitcoin-core/HWI/jobs/664051326

greenknot, 2020-03-20:
I managed to reproduce the issue. Running speculos with a screen (ie. without `--display headless`) allows to notice that each time the app is stuck, some validation from the user is expected (for instance, an address validation). The tests resume once the the validation step is approved using the buttons.

achow101, 2020-03-20:
We work around that by applying a [patch](https://github.com/bitcoin-core/HWI/blob/master/test/data/speculos-auto-button.patch) to speculos that will send the button presses for us depending on what is being displayed.

From all of the travis runs I have done, it seems like it usually gets stuck at startup when we are trying to figure out whether it is started and ready.

greenknot, 2020-03-20:
I think there is a bug in the patch. I encounter the following stacktrace because `seen_msg_hash` isn't always initialized:

```
Traceback (most recent call last):
  File "/home/user/speculos/mcu/screen.py", line 67, in <lambda>
    n.activated.connect(lambda s: klass.can_read(s, self))
  File "/home/user/speculos/mcu/seproxyhal.py", line 167, in can_read
    if self.seen_msg_hash:
AttributeError: 'SeProxyHal' object has no attribute 'seen_msg_hash'
```

achow101, 2020-03-20:
Huh. Our tests must be running in an order that doesn't hit that. We have travis runs that pass and don't time out, and others that do, so this itself isn't the issue (although it is *an* issue).

greenknot, 2020-03-23:
After some investigation, I think there are 2 separate bugs... Could you give PR #41 a try please? Thanks!

achow101, 2020-03-23:
#41 seems to have fixed the issue. I've done a few travis builds and have been running it continuously on my own machine and haven't noticed any timeouts yet.

greenknot, 2020-03-24:
The PR is merged into master (075b43edade91d39b9a7c7673d51790349ac9441). Thanks!

closed 2020-03-24 by greenknot