LedgerHQ/speculos #40: HID support via UHID

Ericson2314, 2020-03-19:
See https://www.kernel.org/doc/Documentation/hid/uhid.txt and https://github.com/torvalds/linux/blob/master/samples/uhid/uhid-example.c

I'll be getting started on this but it would be nice to collaborate with upstream.

greenknot, 2020-03-20:
Hello John,

A few months ago, we actually developed a proof-of-concept code relying on `dummy_usb` to allow usual clients to work with a speculos-like emulator. It isn't super convenient because this kernel module isn't built by default on most Linux distros and there are a few bugs with userland libraries (`libusb` IIRC).

Anyway, here are a few notes and scripts in the [usb branch](https://github.com/LedgerHQ/speculos/commit/5eff94da334cf43674f2505d5f86c7c15670e98d). Disclaimer: this is highly experimental and probably doesn't work as is, but it might give a few hints.

Ericson2314, 2020-03-20:
Awesome! Will check it out.

gitmachtl, 2021-02-15:
Hi, are there any updates available on this? I would like to use speculos for Testing of the Cardano-App with the cardano-hw-cli, but for that speculos must appear like a normal usb device. Is this possible now?

greenknot, 2021-02-16:
Unfortunately no.

I guess you can modify the cardano-hw-cli to communicate with speculos through TCP though.

mullroy, 2023-06-10:
Good day,
I've managed to build the dummy_hcd kernel module and get it loaded, as per your instructions in the USB branch.
I can run the 'setup.sh start' script to create the emulated Ledger USB device. If I run 'lsmod' and 'dmesg' I can see the device:

# lsmod
Bus 003 Device 002: ID 2c97:0001 Ledger Nano S

# dmesg
[] dummy_hcd dummy_hcd.0: USB Host+Gadget Emulator, driver 02 May 2005
[] dummy_hcd dummy_hcd.0: Dummy host controller
[] dummy_hcd dummy_hcd.0: new USB bus registered, assigned bus number 3
[] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.15
[] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[] usb usb3: Product: Dummy host controller
[] usb usb3: Manufacturer: Linux 5.15.0-73-generic dummy_hcd
[] usb usb3: SerialNumber: dummy_hcd.0
[] hub 3-0:1.0: USB hub found
[] hub 3-0:1.0: 1 port detected

# cat /sys/kernel/debug/usb/devices
T:  Bus=03 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=2c97 ProdID=0001 Rev= 2.00
S:  Manufacturer=Ledger
S:  Product=Nano S
S:  SerialNumber=0001
C:* #Ifs= 2 Cfg#= 1 Atr=80 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=01 Prot=01 Driver=(none)
E:  Ad=81(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
E:  Ad=02(O) Atr=03(Int.) MxPS=  64 Ivl=1ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=03(HID  ) Sub=01 Prot=01 Driver=usbhid
E:  Ad=85(I) Atr=03(Int.) MxPS=  64 Ivl=1ms
E:  Ad=07(O) Atr=03(Int.) MxPS=  64 Ivl=1ms

I'm running a Speculos instance with BTC binary, using the default LEDGER PROXY: 127.0.0.1:9999
My speculos instance responds correctly to requests using apduboy or ledgerctl

I've edited the usb-proxy.sh script to use port 9999 as default instead of 1237
If I run usb-proxy.sh it connects to Speculos. If Speculos isn't running, it gives a connection error:
'Connection refused'
There is no further indication of communication between usb-proxy.sh or the running Speculos instance.

I've tried to connect to the USB emulated Ledger device using the Electrum wallet. Latest version 4.4.4 simply
hangs when selecting 'Use a hardware device' when I try to create a new wallet.
With ver 4.3.2 the commandline output shows that it is trying to scan the USB bus for devices, but it says "No hardware device detected."

There's no output on the usb-proxy.py console to indicate that a connection attempt was made.

Can you please provide some hints as to troubleshooting why the Electrum wallet doesn't recognise the emulated device?