LedgerHQ/speculos #54: SDK 1.5.5

mul1sh, 2020-04-20:
Hey guys thanks for this project, I truly appreciate the great work. I was wondering is there anyway to launch speculos for sdk 1.5.5? Because when I try to do this i get the following error
```sh
launcher: invalid SDK version
vnc_server: readall: connection closed
```

And the docker command i'm using to launch speculos is the following

```
docker run --name spec  --rm -it -v "$(pwd)"/.:/speculos/apps -p 1234:1234 -p 40000:40000 -p 41000:41000 -p 42000:42000 speculos --model nanos "./apps/bin/app.elf" --sdk 1.5.5 --seed "secret" --display headless --apdu-port 40000 --vnc-port 41000 --vnc-password toto --button-port 42000
```

Thanks

mul1sh, 2020-04-21:
Getting the same issue with sdk 1.5.  where it tries to start speculos and fails with a syscall
```
[*] using SDK version 1.5
[*] patching svc instruction at 0x4000161a
[*] patching svc instruction at 0x400030a0
launcher: failed to emulate syscall 0x60008381
vnc_server: readall: connection closed
```

greenknot, 2020-04-21:
Sorry, there is not support for SDK 1.5.5 (available SDK versions are 1.6 and 1.5). 

Is it possible to build your app against SDK 1.6, which is the last SDK?

mul1sh, 2020-04-21:
@greenknot sure I'll try with 1.6 but with 1.5 its crashing with this [error](https://github.com/LedgerHQ/speculos/issues/54#issuecomment-616895562) any idea why that might be?

greenknot, 2020-04-21:
`0x60008381` is `io_seph_send` syscall ID for 1.5 and 1.6. While the syscall is [supported for SDK 1.6](https://github.com/LedgerHQ/speculos/blob/66c4cbd62c8d16e403d8b7c6194b6ce2f5af899b/src/emulate_1.6.c#L34-L37), it isn't for SDK 1.5. I guess that's due to `io_seproxyhal_spi_send` being renamed to `io_seph_send` at some point (in 1.6 I think). I don't recall having seen a single app built against SDK 1.5 making use of `io_seph_send`.

mul1sh, 2020-04-21:
Ahh ok I understand now, sdk 1.5 is also failing with the default app  `btc.elf` too so I guess i'll just stick to sdk 1.6 then. Thanks for the clarification.

closed 2020-04-21 by mul1sh