LedgerHQ/speculos #60: Unaligned accesses do not result in errors

jleni, 2020-05-05:
Dereferencing a misaligned pointer results in the device freezing, however, speculos does not emulate correctly this behavior and executes without any problems.

This means that some issues go undetected in speculos and only fail in the field.
It would be good to enable QEMU checks for unaligned access. 

Reference:
http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.faqs/ka15414.html

greenknot, 2020-05-05:
The question is, can QEMU detect unaligned memory access? Unfortunately no if IIUC: there are only 2 ARM instructions raising `EXCP_UNALIGNED`, [vlstm](https://github.com/qemu/qemu/blob/7fbc6a403a0aab834e764fa61d81ed8586cfe352/target/arm/m_helper.c#L1053) and [vlldm](https://github.com/qemu/qemu/blob/7fbc6a403a0aab834e764fa61d81ed8586cfe352/target/arm/m_helper.c#L1123).

So I guess the doc should be updated to mention this speculos limitation.

jleni, 2020-05-06:
Yes, it seems QEMU cannot do it
https://stackoverflow.com/a/51536089/157726

greenknot, 2020-05-07:
Yeah, we noticed that comment too but since it was a few years old, we wanted to be sure it was still accurate ;) I updated (2676c0f66db68a0185c4c9ba6f57f1b872eb2829) the README to mention this limitation.

closed 2020-05-07 by greenknot