LedgerHQ/speculos #61: Button Handling OSX

mul1sh, 2020-05-05:
It seems in OSX speculos doesn't handle the `BUTTON_EVT_RELEASED | BUTTON_LEFT | BUTTON_RIGHT` event when I press down both the `left` and `right` keyboard keys. 

Is there anyway to fix this or to emulate this event programatically? Thanks.

greenknot, 2020-05-06:
@mul1sh does #62 fix the issue?

mul1sh, 2020-05-06:
@greenknot unfortunately no as explained in the PR. i'm also using the `headless` option with docker, not sure if that can be an issue too.

greenknot, 2020-05-07:
Thanks for the [clarification](https://github.com/LedgerHQ/speculos/pull/62#issuecomment-625079760)! Indeed, it looks like a bug in speculos, related to this [io_seproxyhal_button_push()](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/os_io_seproxyhal.c#L946-L948).

mul1sh, 2020-05-07:
@greenknot no worries, glad I could help. Thanks.

greenknot, 2020-05-07:
As mentioned by @mul1sh, the bug can be reproduced on the sia app (it isn't related to OS X and can be reproduced on Linux too), for instance in the [ui_getPublicKey_compare_button()](https://github.com/LedgerHQ/app-sia/blob/22a955bd2dcc2820b4d781ea988d664de1cd657a/src/getPublicKey.c#L58-L85) callback with the following APDU:

```console
echo 'e002000000' | LEDGER_PROXY_ADDRESS=127.0.0.1 LEDGER_PROXY_PORT=9999 ledgerctl send -
```

When a button is pushed (either left or right), `UX_REDISPLAY()` is called, then `UX_REDISPLAY_IDX()` and eventually [io_seproxyhal_init_button()](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/lib_ux/include/ux.h#L428). `io_seproxyhal_init_button()` sets [G_ux_os.button_mask](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/os_io_seproxyhal.c#L416) to `0`.

When the button is released, the callback (`ui_getPublicKey_compare_button()`) is called with `button_mask` being equal to `BUTTON_EVT_RELEASED`, because:

- `G_ux_os.button_mask = 0` because of `io_seproxyhal_init_button()`
- `button_mask = G_ux_os.button_mask | new_button_mask;` ([L910](https://github.com/LedgerHQ/nanos-secure-sdk/blob/1f2706941b68d897622f75407a868b60eb2be8d7/src/os_io_seproxyhal.c#L910))

The condition [BUTTON_EVT_RELEASED | BUTTON_LEFT | BUTTON_RIGHT](https://github.com/LedgerHQ/app-sia/blob/22a955bd2dcc2820b4d781ea988d664de1cd657a/src/getPublicKey.c#L79) is thus never met.

I guess it works on a real device because of some race condition?

greenknot, 2020-05-14:
I've really now idea how it can work on a real device and it's indeed a bug in speculos since the behavior differs. That being said, the sia app uses some legacy UX primitives and I guess this behavior can be fixed (in the sia app) using what's called flow now IIRC, or more conventional button callbacks.

I don't plan to spend more time on this issue, feel free to re-open it if these suggestions don't work for you.

closed 2020-05-14 by greenknot

mul1sh, 2020-05-14:
@greenknot no worries and thanks for the feedback. I think i'll just improvise with other button callbacks moving forward.

Also just out of curiosity what do you mean legacy UX primitives, sounds like the security risk no? Since even the official docs recommend to start developing from the sia app.

PR #62 (2020-05-07): map down key as a shortcut for pressing both left and right keys
Fix #61.