LedgerHQ/speculos #63: Syscall cx_ecfp_scalar_mult is not implemented

niooss-ledger, 2020-05-12:
Hello,

When launching the Monero application (https://github.com/LedgerHQ/app-monero) with Speculos, I get:
```
[*] using SDK version 1.6
[*] patching svc instruction at 0x40004f8a
[*] patching svc instruction at 0x40005754
launcher: failed to emulate syscall 0x60002cf3
```

`0x60002cf3` is `SYSCALL_cx_ecfp_scalar_mult_ID_IN`: syscall `cx_ecfp_scalar_mult` is missing. It is used in Monero for almost every operation on Ed25519, and for example in https://github.com/LedgerHQ/app-monero/blob/ad59fa92bac575962c8e89613c6111802c868ca2/src/monero_crypto.c#L610-L618

It would be nice to support this operation :)

niooss-ledger, 2020-05-13:
The main issue with adding point operations to Ed25519 curve is that OpenSSL does not expose the raw operations. Functions like ``ED25519_sign``, ``ED25519_verify`` and ``ED25519_public_from_private`` all use static functions in https://github.com/openssl/openssl/blob/OpenSSL_1_1_1g/crypto/ec/curve25519.c that are not exporeted.

Therefore implementing ``cx_ecfp_scalar_mult`` requires either using another library that implements Ed25519 operations, or copy/pasting OpenSSL code in a proper way (if OpenSSL's license allows it). What are your thoughts on this?

niooss-ledger, 2020-05-13:
Some other suggestions that do not work:

* Using `EC_KEY *key = EC_KEY_new_by_curve_name(NID_X25519);` and OpenSSL operations. This does not work (this function returns `NULL`, because OpenSSL did not implement X25519 key exchange using its low-level EC API)
* Using OpenSSL's `X25519()` (that performs a scalar multiplication). But converting points from Ed25519 to Curve25519 still needs to be done, and it not trivial.
* Using NaCl, as it only provides a high-level API to Ed25519 (like OpenSSL)

A suggestion that could work:

* Copy a reference implementation like OpenSSL's one or SUPERCOP's one, which is actually what Monero did in https://github.com/monero-project/monero/tree/77a008f71454ce4dd4bce033bc0319a49e2cec51/src/crypto/crypto_ops_builder/ :
  - https://github.com/monero-project/monero/tree/77a008f71454ce4dd4bce033bc0319a49e2cec51/src/crypto/crypto_ops_builder/ref10 contains the reference implementation
  - https://github.com/monero-project/monero/tree/77a008f71454ce4dd4bce033bc0319a49e2cec51/src/crypto/crypto_ops_builder/ref10CommentedCombined contains the same implementation, with horrible Python scripts that glue the code to Monero (like https://github.com/monero-project/monero/blob/77a008f71454ce4dd4bce033bc0319a49e2cec51/src/crypto/crypto_ops_builder/ref10CommentedCombined/test.py and https://github.com/monero-project/monero/blob/77a008f71454ce4dd4bce033bc0319a49e2cec51/src/crypto/crypto_ops_builder/ref10CommentedCombined/MakeCryptoOps.py#L144-L147 -- have you ever seen Python code that calls `awk`, `grep` and `sed` while concatenating strings into `os.system()`? Now you have.)

greenknot, 2020-05-20:
FTR, according to the following commands, other syscalls are missing:

```
monero-app$ ledger-app-monero (master)$ grep 'SVC_Call(' debug/app.asm  | cut -d '(' -f 2 | cut -d ',' -f 1 | sed 's/^SYSCALL_//' | sed 's/_ID_IN$//' | sort -u > /tmp/1
speculos  $ grep -r 'SYSCALL.*(.*' src/ | grep -v define | cut -d '(' -f 2 | cut -d ',' -f 1 | cut -d ')' -f 1 | sort -u >/tmp/2
          $ comm -2 -3 /tmp/1 /tmp/2 | while read l; do grep "<$l>:" debug/app.asm; done | cut -d '<' -f 2 | cut -d '>' -f 1
cx_ecfp_add_point
cx_edward_compress_point
cx_edward_decompress_point
cx_math_invprimem
cx_math_powm
cx_math_sub
cx_math_subm
halt
os_perso_seed_cookie
```

greenknot, 2020-05-25:
I hope that every missing syscall is now supported (at least for ed25519, and `os_perso_seed_cookie` and  `halt` aren't used by the Monero app.) The Monero app now launches successfully and the command `echo '0321000011' |  LEDGER_PROXY_ADDRESS=127.0.0.1 LEDGER_PROXY_PORT=1237 ledgerctl send -` displays an address.

Feel free to re-open this issue if I'm wrong.

closed 2020-05-25 by greenknot