LedgerHQ/speculos #93: bad signature on the TRON app

gre, 2020-08-17:
It seems that, roughly 1 transaction out of 3 creates a bad signature when using the TRON app.

Here is an example of transaction that failed in our ledger-live bot:

```
all accounts sync in 4.4s
▬ Tron 0.2.2 on nanoS 1.6.0
→ FROM Tron 4: TRX 39.8 (54ops) (TBtPherJfuSMaJnWVtuuo2AvjgdfxzUhAb on 44'/195'/3'/0/0) #3 
max spendable ~26.85
★ using mutation 'send max to another account'
→ TO Tron 1: TRX 4.511 (60ops) (TJz8r6opgQn7CXW2pDwfxUj5GiQWHkfB1z on 44'/195'/0'/0/0) #0 
✔️ transaction 
SEND MAX 
TO TJz8r6opgQn7CXW2pDwfxUj5GiQWHkfB1z
STATUS (2076ms)
  amount: TRX 26.851715
  estimated fees: TRX 0
  total spent: TRX 26.851715
✔️ has been signed! (4.1s) {"operation":{"id":"js:2:tron:TBtPherJfuSMaJnWVtuuo2AvjgdfxzUhAb:-3ff0dcd16ca5bf6298c2babc1f1c29c5c644449a3d45c80c620b3e81516fd1c9-OUT","hash":"3ff0dcd16ca5bf6298c2babc1f1c29c5c644449a3d45c80c620b3e81516fd1c9","type":"OUT","blockHash":null,"blockHeight":null,"senders":["TBtPherJfuSMaJnWVtuuo2AvjgdfxzUhAb"],"recipients":["TJz8r6opgQn7CXW2pDwfxUj5GiQWHkfB1z"],"accountId":"js:2:tron:TBtPherJfuSMaJnWVtuuo2AvjgdfxzUhAb:","extra":{},"date":"2020-08-17T10:51:36.828Z","value":"26851715","fee":"0"},"signature":"8f80e965d007c137e4559ade81dfa5d7bf9787dc5873db2efdc8ac0609c20b2d11379a0155969709120c51af4f7186abbb3385671ab6c4d37526b16e3c92b96a00","expirationDate":null,"signatureRaw":{"contract":[{"parameter":{"value":{"amount":26851715,"owner_address":"4115071a84c2d7748e0c29efc03f6035d46b1cf0c0","to_address":"4162e6147e8ff558f579d9dea6031b3468b2b3e4a3"},"type_url":"type.googleapis.com/protocol.TransferContract"},"type":"TransferContract"}],"ref_block_bytes":"c2c1","ref_block_hash":"eda7cac306318b98","expiration":1597661553000,"timestamp":1597661493685}}
⚠️ Error

Error
```

when looking at the technical error, we see the API is rejecting us with 
```
{ "code": "SIGERROR", message: "validate signature error 48c2beb032b1350f3f11f8d73498d953a1713404b77cffcc319b8391fa604a867288e71ff48a1e29c412947e122fc15c281da98b6bb90d2d27a077ef714fe3f600 is signed by TWPLHq8aYcbzEV3c472tjJFYFQ98MYDQuY but it is not contained of permission." }
```

which correspond to signing with an incorrect private key

with the same codebase of live-common we never see this issue with a real device so we highly suspect something on speculos.

niooss-ledger, 2020-08-17:
Tron is using recoverable ECDSA signatures: https://github.com/LedgerHQ/app-tron/blob/v0.3.0/src/helpers.c#L86-L103. This kind of signatures enables recovering the public key from a transaction and its ECDSA signature, but requires information about the parity of the "y" coordinate of the signature point and another parameter more complex to explain (for each ECDSA signatures, there can be at most 4 public keys that generated it and the "recovery ID byte" is used to select one of them).

The implementation of recoverable ECDSA signatures in the Tron application relies on using `info` in the call of `cx_ecdsa_sign`:
```c
    cx_ecdsa_sign(&privateKey, CX_RND_RFC6979 | CX_LAST, CX_SHA256,
                      transactionContext->hash, sizeof(transactionContext->hash),
                      signature, sizeof(signature), &info);
/* ... */
    transactionContext->signature[64] = 0x00;
    if (info & CX_ECCINFO_PARITY_ODD) {
        transactionContext->signature[64] |= 0x01;
    }
```

Currently speculos does NOT fill `info`, as shown by the `UNUSED` macro in https://github.com/LedgerHQ/speculos/blob/ac7ba2246500a1697882b0bc72bee0917f57b519/src/cx_ec.c#L764

In the test output, the signature was `8f80e965d007c137e4559ade81dfa5d7bf9787dc5873db2efdc8ac0609c20b2d11379a0155969709120c51af4f7186abbb3385671ab6c4d37526b16e3c92b96a00`. It would be nice to check whether the same transaction with `01` instead of `00` as last byte, would be fine. If yes, the issue is definitively caused by Speculos not filling `info`. Otherwise, it is something else.

jleni, 2020-09-08:
We are having exactly the same issue in another app:
1- info does not contain correct data
2- even when using RFC6979, signatures change all the time. They are all valid but not stable.

gre, 2020-09-08:
regarding the initial task, i think we can assume it is fixed btw. this issue can be closed, TRON is working now. Maybe worth creating another issue @jleni  if latest speculos is still broken. thanks.

closed 2020-09-08 by gre

jleni, 2020-09-08:
no, adding this PR it works well now :100:

gre, 2020-09-09:
glad to hear that 👍

fullstackdev0410, 2022-06-18:
I got the same error when sending USDT token.
How can I fix this problem?
{
  error: 'SIGERROR',
  message: 'Validate signature error: e13eaebfab8b0ca308fc9aaa970b3418f2d387adc44a0a46a22145a021caf5a4e1223ec0fb7895bd8f5ac8b624930e9ca52cdedafe11e0993dc15b0dc5df69a801 is signed by TGEbBaHpoVXbLzbo1heSENn2GjThAFgT2n but it is not contained of permission.'
}

epheph, 2022-06-24:
@fullstackdev0410 what is up with this `TGEbBaHpoVXbLzbo1heSENn2GjThAFgT2n` ? Did you find its private key somewhere or something? I see many users attempting to recover USDT from this account

Madhav12, 2022-07-10:
> I got the same error when sending USDT token. How can I fix this problem? { error: 'SIGERROR', message: 'Validate signature error: [...]

Hello can you tell me about it my telegram id is https://t.me/njratmy0101 pls contact my Tele so i can try to help see if i can help with your problem pls contact immediately... I think i have solution for your problem if it works..

Madhav12, 2022-07-10:
@cfranceschi-ledger hello are you there pls reply

Madhav12, 2022-07-10:
@fullstackdev0410 hello are u there pls reply

epheph, 2022-07-11:
These are scammers @spacessx1

spacessx1, 2022-07-11:
@epheph I got problem on my own trustwallet

epheph, 2022-07-11:
Got stuck USDT?

spacessx1, 2022-07-11:
Yes and the problem is almost the same dude how can i revoke the signature

epheph, 2022-07-11:
How much USDT

spacessx1, 2022-07-11:
Around 960

epheph, 2022-07-11:
Address? Is it yours or did you find a private key somewhere?

spacessx1, 2022-07-11:
They scammed my trons i got their trustwallet cant get my money back:/ can u write to my tele??

Madhav12, 2022-07-11:
Hello i can help i have solution for this problem send me a pm to my tele @njratmy0101

epheph, 2022-07-11:
You should not solicit or receive help on TG. you will be scammed again.
You can look up flashbots and find a way to request help there from a place you already trust