LedgerHQ/speculos #97: seccomp prevents vnc feature from working

defunctzombie, 2020-09-12:
Using the latest version of master (14ce91a132089d6f0182ab833dc668bdd738c926) and building the docker container (on macOS) results in the vnc feature not working - the vnc server launches however the bridge doesn't send any data back to the python wrapper.

If I DISABLE_SANDBOX in `vnc_server.c` and build again the vnc feature works as expected.

I have not tracked down which calls are being filtered as I am not familiar at all with the internals of how seccomp is meant to work. It does feel tho that using seccomp here may be a bit overkill for what is a testing tool. Maybe this is macOS specific (even tho I am running in docker) and linux users are not seeing the same issues.

nabil-brn, 2020-09-16:
I can confirm this issue is still present on latest version 4211cbc.

```
Traceback (most recent call last):
  File "./speculos.py", line 248, in <module>
    vnc = VNC(args.vnc_port, args.model, args.vnc_password)
  File "/speculos/mcu/vnc.py", line 38, in __init__
Error while loading ./build/src/launcher: No such file or directory
    self.p = subprocess.Popen(cmd, stdin=subprocess.PIPE, stdout=subprocess.PIPE)
  File "/usr/local/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/speculos/mcu/../build/vnc/vnc_server'
```

greenknot, 2020-09-16:
@defunctzombie I fail at reproducing this issue.

You can simply `strace` the process to retrieve the last syscall. For instance, with `bind` commented out (`//ALLOW_SYSCALL(bind)`):
```
$ strace -ff ./build/vnc/vnc_server -v -- -rfbport 5900
...
bind(3, {sa_family=AF_INET, sin_port=htons(5900), sin_addr=inet_addr("0.0.0.0")}, 16) = ?
+++ killed by SIGSYS (core dumped) +++
Bad system call (core dumped)
```

Can you also apply this patch https://github.com/LedgerHQ/speculos/commit/9b051d59d3479b08dcd5dc7151c80497f82490cf to display the syscall number. The expected output should be something like:
```
$ ./speculos.py ./apps/btc.elf --vnc-port 5900
vnc: forbidden syscall (4) in seccomp filter
[*] using SDK version 1.6
[*] patching svc instruction at 0x4000490a
[*] patching svc instruction at 0x40005324
15:59:13.355:vnc: connection with vnc stdout closed
```

Thanks

defunctzombie, 2020-09-16:
@greenknot Thanks for the tips on how to identify the bad syscall. I used the strace approach and got the following:

```
gettimeofday( <unfinished ...>)         = ?
+++ killed by SIGSYS +++
Bad system call
```

The rough steps:

* Build speculos master branch in a dockerfile
* Run the docker image exposing port 8001
* Run `strace -ff ./build/vnc/vnc_server -v -- -rfbport 8001` within the container
* Connect to `localhost:8001` from my MacOS host using VncViewer
* vnc_server immediately crashes in the container with the above SIGSYS

defunctzombie, 2020-09-16:
Adding `ALLOW_SYSCALL(gettimeofday)` prevented vnc from crashing when I connect with VncViewer.

commit 5d2b3fad (2020-09-16): vnc: add gettimeofday to the seccomp filter

Fix #97.
PR #101 (2020-09-16): vnc: add gettimeofday to the seccomp filter
Fix #97.
closed 2020-09-16 by greenknot