LedgerHQ/speculos #100: docker-compose up doesn't work

chrisbward, 2020-09-15:
go ahead, try it

chrisbward, 2020-09-15:
```
Successfully built bc243599b818
Successfully tagged speculos:latest
Starting speculos_nanos_1 ... done
Attaching to speculos_nanos_1
nanos_1  | Traceback (most recent call last):
nanos_1  |   File "./speculos.py", line 248, in <module>
nanos_1  |     vnc = VNC(args.vnc_port, args.model, args.vnc_password)
nanos_1  |   File "/speculos/mcu/vnc.py", line 38, in __init__
nanos_1  |     self.p = subprocess.Popen(cmd, stdin=subprocess.PIPE, stdout=subprocess.PIPE)
nanos_1  |   File "/usr/local/lib/python3.8/subprocess.py", line 854, in __init__
nanos_1  |     self._execute_child(args, executable, preexec_fn, close_fds,
nanos_1  |   File "/usr/local/lib/python3.8/subprocess.py", line 1702, in _execute_child
nanos_1  |     raise child_exception_type(errno_num, err_msg, err_filename)
nanos_1  | FileNotFoundError: [Errno 2] No such file or directory: '/speculos/mcu/../build/vnc/vnc_server'
```

greenknot, 2020-09-16:
It seems that the (optional) VNC server isn't built. The following [command line](https://github.com/LedgerHQ/speculos/blob/master/doc/build.md#vnc-support-optional) shows how to build it.

I'm not familiar with docker-compose though. Is it related to speculos [Dockerfile](https://github.com/LedgerHQ/speculos/blob/master/Dockerfile)/[doc](https://github.com/LedgerHQ/speculos/blob/master/doc/docker.md)?

nabil-brn, 2020-09-16:
I can confirm this issue is still present on latest version 4211cbc.

```
Traceback (most recent call last):
  File "./speculos.py", line 248, in <module>
    vnc = VNC(args.vnc_port, args.model, args.vnc_password)
  File "/speculos/mcu/vnc.py", line 38, in __init__
Error while loading ./build/src/launcher: No such file or directory
    self.p = subprocess.Popen(cmd, stdin=subprocess.PIPE, stdout=subprocess.PIPE)
  File "/usr/local/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/speculos/mcu/../build/vnc/vnc_server'
```

nbourenane@LMPS0002 ~/misc/speculos (git)-[master] % docker images
REPOSITORY          TAG                 IMAGE ID            CREATED             SIZE
speculos            latest              933a44f05e94        28 minutes ago      445MB
ledgerhq/speculos   latest              c5b40b5785b2        20 hours ago        487MB
```
---
4211cbc (HEAD -> master, me/master, me/HEAD, ledger/master) Merge pull request #99 from LedgerHQ/glyphs_fix

nabil-brn, 2020-11-17:
https://github.com/LedgerHQ/speculos/pull/109

closed 2020-11-17 by nabil-brn