LedgerHQ/speculos #102: cx_eddsa_get_public_key: invalid key size with value 64, but works on physical device

sargue, 2020-09-17:
Hi,

I'm seeing a different behavior between speculos and a physical Nano S.

I tried speculos to run this app (https://github.com/jelurida-dev/app-ledger-ardor-main) and got this error message:

```
[*] using SDK version 1.6
[*] patching svc instruction at 0x4000637a
[*] patching svc instruction at 0x400073b8
09:50:31.254:apdu: > e0060100142c00008072410080000000800100008000000000
09:50:31.255:seproxyhal: printf:
09:50:31.257:seproxyhal: printf:  canery check 1 last command number 0
09:50:32.376:seproxyhal.ticker: skipping ticker events (expected within a debugger)
launcher: cx_eddsa_get_public_key: invalid key size (0x40)
vnc_server: readall: connection closed
```

The emulator code strictly expects a 32 in the private length: https://github.com/LedgerHQ/speculos/blob/master/src/cx_ec.c#L288-L291

Just checked our code, done by a previous developer. Funny comment...
```
// weird custom initilization, code copied from Cardano's EdDSA implementaion
privateKey.curve = CX_CURVE_Ed25519;
privateKey.d_len = 64; //don't know why the length is 64 instead of 32, it just works
os_memmove(privateKey.d, KLKR, 32); //Copy just the KL part
```
"it just works"...
And indeed it does on the Nano S. :shrug:

Saltari, 2021-08-10:
With the integration of the 2.0 SDK speculos now embeds the crypto binary of the 2.0 version (with most of the crypto direct calls)
Therefore all the EDDSA functions are now the same between a real Nano S and speculos
The problem remains on LNX (1.3/1.2) but will disappear soon with the next firmware update.

sargue, 2021-08-10:
I confirm that the latest speculos using SDK 2.0 indeed works as expected.

closed 2021-08-10 by sargue