LedgerHQ/speculos #110: Build fails error "multiple definition of `G_cx'; libemu.a(cx_blake2b.c.o):(.bss+0x0): first defined here"

lovesh, 2020-11-18:
I am building on Kubuntu 20.10 with kernel 5.8.0-29-generic and building the master branch of the repo. I followed the instructions [here](https://github.com/LedgerHQ/speculos/blob/master/doc/build.md), installed the dependencies with apt and then ran

```
cmake -Bbuild -H. -DWITH_VNC=1
make -C build/
```

I get the following errors
```
/usr/lib/gcc-cross/arm-linux-gnueabihf/10/../../../../arm-linux-gnueabihf/bin/ld: libemu.a(cx_hmac.c.o):(.bss+0x0): multiple definition of `G_cx'; libemu.a(cx_blake2b.c.o):(.bss+0x0): first defined here
/usr/lib/gcc-cross/arm-linux-gnueabihf/10/../../../../arm-linux-gnueabihf/bin/ld: libemu.a(cx_sha256.c.o):(.bss+0x0): multiple definition of `G_cx'; libemu.a(cx_blake2b.c.o):(.bss+0x0): first defined here
/usr/lib/gcc-cross/arm-linux-gnueabihf/10/../../../../arm-linux-gnueabihf/bin/ld: libemu.a(cx_sha512.c.o):(.bss+0x0): multiple definition of `G_cx'; libemu.a(cx_blake2b.c.o):(.bss+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/launcher.dir/build.make:85: src/launcher] Error 1
```

Even if i try to build without VNC, i get similar errors.

greenknot, 2020-11-19:
#112 should fix this issue (while I didn't test it on Kubuntu 20.20). Please re-open it if I'm wrong.

closed 2020-11-19 by greenknot

lovesh, 2020-11-27:
Thanks. Works.