LedgerHQ/speculos #111: Cannot setup device when running with Docker

lovesh, 2020-11-18:
I wanted to use the emulator with Ledger live. Used Docker and the instructions [here](https://github.com/LedgerHQ/speculos/blob/master/doc/docker.md) to run. I can connect to emulator using VNC and also see the UI in console when running with `text` display. I can navigate with left, right, up, and down arrows but I cannot setup a new device by pressing both left and right keys together as Ledger Live says.

greenknot, 2020-11-19:
There are only 2 buttons on the Ledger Nano S, which are accessible with `←` and `→` keys ;) You can use the `↓` key as a shortcut for pressing both `←` and `→` keys.

closed 2020-11-19 by greenknot

lovesh, 2020-11-27:
I can use down arrow. Is it possible to setup the emulator like a ledger device and connect with ledger live?