LedgerHQ/speculos #121: Automation HTTP service

aguinet, 2021-01-04:
It could be interesting to have Speculos creating an HTTP service that would serve an API which would allow to control automation (and the device itself) while it's running. This would potentially replace also the features provided by `--automation-port`, `--button-port` and `--finger-port`.

One use case I have in mind is being able to test Ledger companion apps on Windows & OSX using e.g. GitHub actions. Indeed, in GitHub actions, for a job, you can run a "background" service under Linux, that would be used by the job itself running on another OS. So, we could have Speculos running under Linux, exposing its automation port, for which we could push "automation JSON" depending on the test case.

This is a first "brain dump", and IMHO needs more thinking, especially on the API definition.

closed 2021-06-16 by greenknot

greenknot, 2021-06-16:
@aguinet done in #170, enjoy!

aguinet, 2021-06-16:
Wow this is very cool! Will test this ASAP :) Thanks @greenknot !