LedgerHQ/speculos #130: Issue with os_global_pin_is_validated with nano x

rllola, 2021-02-03:
I am not quite sure what issue is actually but it seems that I have different behavior between nano s and nano x.

In our app we have a `os_global_pin_is_validated` check. We are using zemu to run test which depend on speculos. The check pass when running the test for nano s but not for nano x.

If someone has an idea of what I am missing that would be nice.

greenknot, 2021-02-03:
Can you share technical details (such as the test source code)?

jleni, 2021-02-04:
Adding a bit more info to this issue (Thanks Raf/Edouard)!

At the moment, `os_global_pin_is_validated` always returns `0xAA`:
https://github.com/LedgerHQ/speculos/blob/master/src/bolos/os_1.6.c#L3

However, in Nano X this value has changed to `0xB0105011`
https://github.com/LedgerHQ/nanox-secure-sdk/blob/a3253dec07fe77b7bbb60f26fd78e4cec37bc02c/include/os.h#L991

so the response does not match the corresponding `BOLOS_UX_OK` in our code and CI fails

jleni, 2021-02-04:
I would create a PR, but I am not sure how to get the current device model in this context 

https://github.com/LedgerHQ/speculos/blob/4e0f563b294235cdd610df3d5df2edb26747a297/src/bolos/os_1.6.c#L13

greenknot, 2021-02-04:
Does PR #131 fix you issue?

rllola, 2021-02-04:
As far as I can tell it doesn't.

I have modified our docker build to integrate your modification by doing `git checkout 23036d118da8d7dd0c6f9c8fd805bf1b382c1846` at this line :
https://github.com/Zondax/docker-builder/blob/master/zemu/Dockerfile#L48

Then I have build and push to my docker hub (https://hub.docker.com/r/rllola/builder-zemu) and updated `@zondax\zemu` to use my image by default `rllola\builder-zemu:latest` on this line : 
https://github.com/Zondax/zemu/blob/master/src/index.js#L54

Then I did `yarn build` and `yarn link` it to the repo...

There is many steps to it and many places wher I can messed up. But if I did everything right it still doesn't validate the pin.

rllola, 2021-02-04:
Could it be that here it should be `sys_os_global_pin_is_validated_1_2` instead of `os_global_pin_is_validated_1_2` ? Is it missing the `sys_` prefix ?

https://github.com/LedgerHQ/speculos/blob/23036d118da8d7dd0c6f9c8fd805bf1b382c1846/src/emulate_1.2.c#L71

greenknot, 2021-02-04:
My bad, `BOLOS_UX_OK` was still defined to an incorrect value. I just force-pushed cb9c723ca2175baa66eccbf381ade054a73b2f2c, it should be good now.

jleni, 2021-02-05:
Thanks for fixing this! What's the ETA for merging the PR?
Or you prefer us to verify and confirm?

greenknot, 2021-02-05:
I just merged it. Please re-open the issue if there's something wrong.

commit cb9c723c (2021-02-04): fix a few Nano X syscalls

Fixes #130
PR #131 (2021-02-05): fix a few Nano X syscalls
Fixes #130
closed 2021-02-05 by greenknot