LedgerHQ/speculos #133: Add ability to mock syscalls for tests

bigspider, 2021-02-08:
While writing functional tests, it is often the case that some code paths only happen based on the return values of certain APDUs.
Therefore, to get a good test coverage, it would be very helpful to hook into the syscall emulation and programmatically control the syscalls' return values.

In the best of all possible worlds, python tests would have a `speculos` fixture running an app, that has god-mode control over the device:

```python
def test_something(speculos):
    def mock_os_my_syscall(param):
       return 1 if param == 0xF00 else 0

   speculos.on("os_my_syscall", mock_os_my_syscall)

  # test stuff
``` 

Ideally, it should be possible to alter the behavior _during_ execution, to test more complex interactions where the syscall is called multiple times and its return values is different.

My current use case is to test code depending on the result of `os_global_pin_is_validated`; so even less powerful customization on speculos might be enough.

greenknot, 2021-02-11:
I added a proof-of-concept code in the [hook-syscalls branch](https://github.com/LedgerHQ/speculos/commits/hook-syscalls). An HTTP request is made each time the syscall `os_global_pin_is_validated` is called and the HTTP response is used to set the return value of the syscall. The syscall id id is passed in the URL (`GET /?syscall=0x6000a03c HTTP/1.0`).

As an example, here is an HTTP server wich returns the same value than the [syscall implementation](https://github.com/LedgerHQ/speculos/blob/hook-syscalls/src/bolos/os_1.6.c#L11-L14) (`0xaa` is `BOLOS_UX_OK` and `0x9000a078` is `SYSCALL_os_global_pin_is_validated_ID_OUT`).
```console
cd /tmp/
echo '0xaa 0x9000a078' > index.html
python3 -m http.server
```

```console
./speculos.py --hook-syscalls ./apps/btc.elf
```

Free free to customize that for your needs, ie.:
- develop your own HTTP server
- modify the [hook_syscall](https://github.com/LedgerHQ/speculos/blob/01421bbafd55347336082d4d1c467ab69b2a23e4/src/hook_syscalls.c#L65-L81) function to add mocked syscalls
- modify the HTTP server [host/port/url](https://github.com/LedgerHQ/speculos/blob/01421bbafd55347336082d4d1c467ab69b2a23e4/src/hook_syscalls.c#L19)

Hope it helps!

bigspider, 2021-02-11:
That's amazing, thanks! That's going to be very useful.