LedgerHQ/speculos #134: Unexpected crash for nano x

rllola, 2021-02-09:
Hi,

I had another crash when loading a nano x application.
Here the message :
```
serving at port 9998
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-zondax'
[*] using SDK version 1.2
[*] patching svc instruction at 0x40001e68
[*] patching svc instruction at 0x4000c628
launcher: failed to emulate syscall 0x60009c8b
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
vnc_server: readall: connection closed
```

greenknot, 2021-02-10:
This is the `os_sched_last_status` syscall of the SDK 1.2:
```console
$ grep -ir 0x60009c8b sdk/
sdk/bolos_syscalls_1.6.h:#define SYSCALL_os_sched_last_status_ID_IN  0x60009c8bUL
```

I modified the Bitcoin app to call this syscall:
```diff
diff --git a/src/main.c b/src/main.c
index f8ddf1e..777572e 100644
--- a/src/main.c
+++ b/src/main.c
@@ -1329,6 +1329,8 @@ __attribute__((section(".boot"))) int main(int arg0) {
     // exit critical section
     __asm volatile("cpsie i");
 
+    os_sched_last_status(0);
+
     // ensure exception will work as planned
     os_boot();
 
```

and it works fine. The syscall is successfully executed (as shown by the `--trace` argument):
```console
$ ./speculos.py --model nanox --trace ~/app-bitcoin/bin/app.elf |& head -8
[*] using SDK version 1.2
[*] patching svc instruction at 0x4000173c
[*] patching svc instruction at 0x4000867a
[*] patching svc instruction at 0x40009102
[*] patching svc instruction at 0x40009628
[*] patching svc instruction at 0x40009656
[*] patching svc instruction at 0x40009682
[*] syscall: os_sched_last_status(0) = -1341108207
```

Can you share your app (the .elf file)?

greenknot, 2021-02-22:
@rllola does the crash still happen?

rllola, 2021-02-23:
I believe it has been fixed by one of your developer. Another team member who encountered this issue too contacted one of your engineer via slack.

So this can be closed.

closed 2021-02-23 by rllola