LedgerHQ/speculos #150: Return correct result for getting app name and version (apdu b0 01 00 00)

danimoh, 2021-03-22:
This request currently returns dummy data according to https://github.com/LedgerHQ/speculos/issues/22.
Would it be possible to return the actual data from the loaded app / run the actual `os_io_seproxyhal_get_app_name_and_version` of the loaded app?

greenknot, 2021-03-23:
Unfortunately no, otherwise it would be implemented.

The rationale is that these parameters (name and version, icon, allowed derivation paths, etc.) are given during the app install but aren't embedded in the .elf file itself.

As mentioned in the comments of #22, the `SPECULOS_APPNAME` environment variable allows to set the app name and version (eg. `SPECULOS_APPNAME=blah:1.2.3.4`). I just updated the doc in #151 to mention that.

closed 2021-03-23 by greenknot

danimoh, 2021-03-23:
Thanks for the clarification :+1: