LedgerHQ/speculos #156: Add support for --display text for the Nano X

danimoh, 2021-03-28:
Using `--display text` in combination with `--model nanox` currently throws error `unsupported model 'nanox' with argument -x`.

Would it be possible to add text support for the Nano X?

greenknot, 2021-05-06:
Related PR: #118

dkremer-ledger, 2021-05-11:
Hello @danimoh ,

Thank you for your request. The PR #118 introduces an experimental support for a text output. However, the emulator displays raw bitmap character sent directly by the apdu and a true 'text mode' is not really available at the moment.

I have to clean the PR a bit to adjust this to the existing --text option for nanos.

In the meantime, you can check out the code if you are in that sort of thing and give additional feedback.

An example of how it works at the moment:

```
./speculos.py --trace --m nanox --sdk 1.2 apps/nanox\#boilerplate\#1.2\#00000000.elf --log-level root:DEBUG --record-frame out
```
`out` is the prefix of filenames that will contain frames displayed by the emulator.

Files `out-%i.txt` are the raw bitmap in ascii code. the file `out-text.txt` is a text built from the bitmap chars sent to the screen when they have been mapped to the correct character from the bitmap font availables in the SDK 2.0.

greenknot, 2021-06-16:
I'm not sure why `--display text` was incompatible with `--model nanox`. Anyway, this limitation was removed in 96b509b4e597a7ffb7d914066d8a93acd5f6d8f9.

closed 2021-06-16 by greenknot