LedgerHQ/speculos #175: app-boilerplate failed to emulate syscall

ixje, 2021-06-16:
I'm in the process of going through the developer docs and setting up a dev environment for creating an app. The Ledger [app-boilerplate](https://github.com/LedgerHQ/app-boilerplate) fails to run on Speculos with the following log
```
./speculos.py ../../app-boilerplate/bin/app.elf 
/opt/ledger/speculos/mcu/nanox_ocr.py:155: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if self.text is b"":
[*] speculos launcher revision: 53d114d
[*] using SDK version 1.6
[*] patching svc instruction at 0x40002230
launcher: failed to emulate syscall 0x60010b06
```

The syscall in question is defined in `nanos-secure-sdk` as 
```
include/syscalls.h:#define SYSCALL_try_context_set_ID_IN 0x60010b06UL
```

I hope it is not unrealistic to expect that at least the boilerplate app works, no?

greenknot, 2021-06-16:
I guess your app is built against the version 2.0 of the SDK. By default, speculos uses the version 1.6 of the SDK (as mentioned in the log) and we'll switch to 2.0 by default soon. Meanwhile, could you try the following command-line:

```shell
./speculos.py --sdk 2.0 ../../app-boilerplate/bin/app.elf 
```

greenknot, 2021-06-16:
BTW, feel free to join the #speculos channel on [Ledger's developer slack](https://ledger.readthedocs.io/en/latest/additional/publishing_an_app.html#contact).

ixje, 2021-06-16:
Great, that works!

closed 2021-06-16 by ixje