LedgerHQ/speculos #204: The Nano X OCR incorrectly regognizes uppercase I as a L

greenknot, 2021-07-01:
Here is the output of speculos:

```shell
./speculos.py --log-level automation:DEBUG --model nanox --automation file:/app-xrp/tests/automation.json /app-xrp/bin/app.elf
...
12:38:16.440:automation: getting actions for "lssuer
r9cZA1mLK5R5Am25A
rfXFmqgNwjZgnfk59" (47, 6)
```

(notice `lssuer` instead of `Issuer`.)

For the following APDU:
```shell
$ LEDGER_PROXY_ADDRESS=127.0.0.1 LEDGER_PROXY_PORT=9999 pytest-3 -v -s -k test_sign_valid_tx tests/
$ echo e0040040b4058000002c80000090800000008000000000000000120010228000000024000000192a21fb3df12e0000000150116f1dfd1d0fe8a32e40e1f2c05cf1c15545bab56b617f9c6c2d63a6b704bef59b68400000000000000a69e0c68db7b413ec0000000000000000000000000055534400000000005e7b112523f68d2f5e879db4eac51c6698a6930481140511e17db83bb6f113939d67bc8ea539edc926fc83147148ebebf7304ccdf1676fefcf9734cf1e780826 | LEDGER_PROXY_ADDRESS=127.0.0.1 LEDGER_PROXY_PORT=9999 ledgerctl send -
```

bigspider, 2022-05-30:
It also has problem with the capital "S", which seems to be skipped altogether: `Spend from` on screen generates the text event with `pend from`, and 'TEST 0.00040006' generates the text event `TET 0.00040006`.

haammar-ledger, 2023-05-10:
Same issue encountered when trying to update Live's [bot test framework](https://github.com/LedgerHQ/ledger-live/wiki/LLC:bot) to use NanoSP applications.

xchapron-ledger, 2023-09-04:
Should be fixed with recent OCR changes and mostly https://github.com/LedgerHQ/speculos/pull/418

closed 2023-09-04 by xchapron-ledger