LedgerHQ/speculos #230: Versioning scheme for Speculos package

niooss-ledger, 2021-07-22:
Hello,
As Speculos is now available on [PyPI](https://pypi.org/project/speculos/) and [TestPyPI](https://test.pypi.org/project/speculos/), some decisions have to be made on how versions are defined in Speculos.

For TestPyPI, every commit on `master` is versioned as `maj.min.count` where `maj.min` is the last tag and `count` the number of commits since the last tag. This was implemented in https://github.com/LedgerHQ/speculos/pull/225 and https://github.com/LedgerHQ/speculos/pull/220. This work is/will be integrated with the CI, and is simple enough to understand.

However this does not fit the "semantic versioning" principles. For example when a breaking change is introduced in version 1.0.42 (for example "let's remove feature A for Speculos"), this would automatically create a version 1.0.43 when this change is merged, instead of version 1.1.0 or 2.0.0. (Then a git tag could be added to Speculos to create a proper release).

I see mainly two ways of solving this:

* Only create releases (on PyPI) when a git tag is created. This would prevent a 1.0.43 from being created (and the CI would create such a version on TestPyPI for testing purposes). This would require the project maintainers to regularly tag a new release, which would introduce "more overhead".
* Automatically release versions of `master` branch on PyPI like on TestPyPI and create a way to stop releasing new versions on PyPI when a breaking change occurs (for example the version in `setup.py` could be bumped to 2.0.0 and this it does not match the last git tag, the release is not done). This would introduce more complexity in the version handling process, which could become more fragile.

There could be other ways to fix this. Let's discuss this in this issue!

greenknot, 2021-07-29:
Thanks @niooss-ledger for tackling this topic.

Another idea which doesn't involve tags: just use the version from `setup.py`. I don't know if it's considered good practice, but tags are of no use currently and introduce some complexity in the release process.

The following command-line:

```console
$ git rev-list --count "$(git blame -L$(git grep -h --line-number version= setup.py | cut -d : -f 1),+1 setup.py | cut -d ' ' -f 1)..HEAD"
```

1. Retrieves the line number of `setup.py` which contains `version=` thanks to `git grep`
2. Passes it to `git blame` to get the revision when this line was modified last
3. Eventually get the number of commits since this last revision

niooss-ledger, 2021-07-29:
Your command-line is simple enough to be integrated in a helper script, and I did this in https://github.com/LedgerHQ/speculos/pull/234. This way, we no longer rely on git tags for version pushed in TestPyPI, and this could follow "semantic versioning" principles if Pull Requests which introduce breaking changes also bump the version in `setup.py` :)

Also, with https://github.com/LedgerHQ/speculos/pull/234, a new flow becomes possible: when we want to release Speculos version `x.y.0`:

* We bump the version in `setup.py`, create a PR and merge it, as usual.
* GitHub Actions CI creates a package which is really named `x.y.0` and which is published on TestPyPI.
* We can test installing this test package and verify that things work as expected.
* When ready, we can push a tag `x.y.0` to the repository.
* With this tag, GitHub Actions CI creates a package and publishes it on PyPI (publishing it on TestPyPI is expected to fail at this point, because the package already exists, but this should not matter)

So this would be the first option I suggested in the first message of this issue, with the improvements of using the real `x.y.0` version on TestPyPI.

This procedure seems to be light and intuitive. What do you think?

greenknot, 2021-07-29:
> Your command-line is simple enough to be integrated in a helper script, and I did this in #234. This way, we no longer rely on git tags for version pushed in TestPyPI, and this could follow [...]

Cool :+1: 

> So this would be the first option I suggested in the first message of this issue, with the improvements of using the real `x.y.0` version on TestPyPI.
> 
> This procedure seems to be light and intuitive. What do you think?

What happens if things don't work as expected with a package published as `x.y.0` on TestPyPI? We'll have to fix stuff and:

- The minor version will be incremented by the number `z` of commits which fix the issue. The version which will be tagged and published on PyPi will be `x.y.z`.
- Otherwise bump (again) the version in `setup.py`.

Also, how should we test if a package is ready to be published on PyPI and who is responsible of pushing tags?

I wonder if packages published on TestPyPI could be considered as release candidates. When something looks good, just bump the version in `setup.py` and let the CI publish the package on PyPI? It also allows us to maintain a ChangeLog which could be updated along `setup.py`? Everything can happen and be reviewed on GitHub.

niooss-ledger, 2021-07-29:
In fact, as bumping the version in `setup.py` would be done in Pull Requests, `x.y.0` will never reach PyPI, but at least `x.y.1` (to count the Merge commit)... Except if we merge with rebasing (or without merge commit) Pull Requests which would bump the version. But doing so would not be compatible with "including the version bump in PR which break backward compatibility". So in my humble opinion, we should accept publishing `x.y.z` versions with `z` never being zero, on PyPI.

> Also, how should we test if a package is ready to be published on PyPI and who is responsible of pushing tags?

According to https://packaging.python.org/guides/using-testpypi/, at least by running in a clean environment (like one given by `podman run --rm -ti docker.io/library/python:3.6-slim bash`):

```sh
pip install --index-url https://test.pypi.org/simple/ --extra-index-url https://pypi.org/simple speculos
```

And maybe also launching apps using the newly-installed speculos... It would be nice to have a simple wrapper which would run Speculos tests (from `tests/`) on the installed `speculos` package, if this is possible.

> I wonder if packages published on TestPyPI could be considered as release candidates.

+1 :)

lpascal-ledger, 2023-02-09:
Bumping this thread with retex from app tooling project.

The way we deploy python package are the following:
- We work with two branches, `develop` and `master`.
- The default branch is `develop` and most if not all of the work is done on it, while `master` is only used to hold "stable release" merge commit & tags.
-  Deployments are triggered on `develop` and `master` branches by analyzing the last commit message. If it contains the word "release", a deployment is performed. The release correct version is expected to have been updated in the code.
- `develop` deployment goes to `test.pypi.org`. `master` deployment goes to `pypi.org`.

Speculos branches has already been changed closer to this model: the default branch is no longer `master` but has been renamed `develop`. But except this renaming, the previous deployment process remains unchanged: every merge on `develop` creates a new release `x.y.z` with `z` the number of commits since last version in the `setup.py`.

lpascal-ledger, 2023-02-09:
What could be done from here:

- we keep the current behavior on `develop`, with the `patch` keeps being the number of commits since last version.
- stable release could follow another scheme, where we create a PR from `develop` + one commit updating the version and named `"[release] vX.Y.Z"` to `master`, and the merge push the package to `pypi.org`.

That way, on `test.pypi.org` we got `vX.Y.Z` to be tested, then when deemed OK this version is released as `vX.Y+1.0` or `vX+1.Y.0` on `pypi.org`.
(In fact his latter version should also be deployed on `test.pypi.org` to sync the two package repos).

This adds the need to create a commit between `develop` and `master`, but could be a good place to update CHANGELOG and such.

lpascal-ledger, 2023-02-09:
This proposition is partly implemented in [this PR](https://github.com/LedgerHQ/speculos/pull/342), given the existence of a `PYPI_PUBLIC_API_TOKEN` token.

lpascal-ledger, 2023-05-29:
In order to be able to deploy on `*.pypi.org` repository, which only accept semver version, the `+g<A>` postfix has been removed.

Current behavior is:

- when merging on a main branch (`develop` or `master`) -> deployment on `test.pypi.org` with an autogenerated version number (with the `.dev<number of commits since latest tag>`.
- when a tag is pushed -> deployemnt on `pypi.org`, given the tag matches the lates version documented in the `CHANGELOG.md`

lpascal-ledger, 2023-05-29:
Resolved in #381