LedgerHQ/speculos #277: Mac Screen Sharing Kills VNC Server

JesseKuntz, 2022-03-28:
When using Mac Screen Sharing to open a new VNC connection to Speculos, the server is killed upon entering a password (any password, even the incorrect one).

**Mac info:** 
MacOS Big Sur
11.5.1
Intel chip

**Demo:** 

[image]

**Command:** 

```
docker run -v /Users/jessekuntz/apps/coinapps/:/speculos/apps -p 40001:40000 -p 41001:41000 -p 42001:42000 -p 43001:43000 -p 5000:5000 -e SPECULOS_APPNAME=Mina:1.0.6 --name speculosID-1 ghcr.io/ledgerhq/speculos --model nanos ./apps/nanos/2.1.0/Mina/app_1.0.6.elf --sdk 2.1 --display headless --vnc-password live --apdu-port 40000 --vnc-port 41000 --button-port 42000 --automation-port 43000
```

Using the same arguments that are used in [`ledger-live-common`](https://github.com/LedgerHQ/ledger-live-common) `src/load/speculos.ts` - working on the Mina native integration into Ledger Live.

Additionally, it looks like the web interface isn't available at 127.0.0.1:5000, but that might be some more lack of understanding on my part. Please let me know if any more information would be helpful to diagnose the issue.

greenknot, 2022-03-30:
It might be an issue similar to #97.

Could you try to add `#define DISABLE_SANDBOX` before https://github.com/LedgerHQ/speculos/blob/6bb034916fee75728f82fc81908f4abe053080e8/src/vnc/vnc_server.c#L26 and rebuild speculos?

JesseKuntz, 2022-03-31:
Haven't built it locally yet - I made the change, and then did:
1. `cmake -Bbuild -H. -DWITH_VNC=1`
2. `make -C build/`
3. `docker build ./ -t speculos`
4. `docker image tag docker.io/library/speculos speculos-local`
5. Docker run:
```
docker run -v /Users/jessekuntz/apps/coinapps/:/speculos/apps -p 40001:40000 -p 41001:41000 -p 42001:42000 -p 43001:43000 -p 5000:5000 -e SPECULOS_APPNAME=Mina:1.0.6 speculos-local --model nanos ./apps/nanos/2.1.0/Mina/app_1.0.6.elf --sdk 2.1 --display headless --vnc-password live --apdu-port 40000 --vnc-port 41000 --button-port 42000 --automation-port 43000
```

And then ended up with:
```
Traceback (most recent call last):
  File "/speculos/./speculos.py", line 7, in <module>
    main()
  File "/speculos/speculos/main.py", line 305, in main
    vnc = VNC(args.vnc_port, screen_size, args.vnc_password)
  File "/speculos/speculos/mcu/vnc.py", line 37, in __init__
    self.p = subprocess.Popen(cmd, stdin=subprocess.PIPE, stdout=subprocess.PIPE)
  File "/usr/local/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/local/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/speculos/speculos/mcu/../resources/vnc_server'
```

Wondering if you have any ideas in this case? Thank you!

greenknot, 2022-04-04:
According to the error message `'/speculos/speculos/mcu/../resources/vnc_server'`, the VNC server wasn't built.

In step `3`, I guess the container image should be `speculos-local`.

JesseKuntz, 2022-04-18:
Hrm... on a closer look, it seems that the `make -C build/` step is not successful, because it can't find the `arm-linux-gnueabihf-gcc` command - which seems that it is not available for MacOS, so is it possible to rebuild using MacOS? Or is a Linux environment necessary for development?

greenknot, 2022-05-05:
The docker image [speculos-builder](https://github.com/LedgerHQ/speculos/pkgs/container/speculos-builder) contains all dependencies to build speculos.

closed 2022-07-05 by greenknot