LedgerHQ/speculos #282: Werkzeug >= 2.1.0 triggers issues when using the `SpeculosClient`

lpascal-ledger, 2022-05-02:
I've encountered an unclear bug with a fresh install of Speculos with `pytest`:

```
==================================================================== ERRORS ====================================================================
_______________________________________________________ ERROR at setup of test_app_name ________________________________________________________
tests/speculos/conftest.py:32: in client
    with SpeculosClient(app=str(file_path), args=args) as client:
../../../../.virtualenvs/new/lib/python3.8/site-packages/speculos/client.py:216: in __enter__
    self.start()
../../../../.virtualenvs/new/lib/python3.8/site-packages/speculos/client.py:208: in start
    self.open_stream()
../../../../.virtualenvs/new/lib/python3.8/site-packages/speculos/client.py:73: in open_stream
    check_status_code(self.stream, "/events")
../../../../.virtualenvs/new/lib/python3.8/site-packages/speculos/client.py:30: in check_status_code
    raise ClientException(
E   speculos.client.ClientException: HTTP request on /events failed, status=400, error=b'{"message": "The browser (or proxy) sent a request that this server could not understand."}\n'
------------------------------------------------------------ Captured stdout setup -------------------------------------------------------------
 * Serving Flask app 'speculos.api.api' (lazy loading)
 * Environment: development
 * Debug mode: off
------------------------------------------------------------ Captured stderr setup -------------------------------------------------------------
10:36:41.448:werkzeug:  * Running on all addresses (0.0.0.0)
   WARNING: This is a development server. Do not use it in a production deployment.
 * Running on http://127.0.0.1:5000
 * Running on http://10.0.56.136:5000 (Press CTRL+C to quit)
[*] speculos launcher revision: 
[*] using SDK version 2.1 on nanos
[*] loading CXLIB from "/home/lpascal/.virtualenvs/new/lib/python3.8/site-packages/speculos/cxlib/nanos-cx-2.1.elf"
[*] patching svc instruction at 0x126358
[*] patching svc instruction at 0x400025d8
10:36:41.490:werkzeug: 127.0.0.1 - - [02/May/2022 10:36:41] "GET /events?stream=true HTTP/1.1" 400 -
-------------------------------------------------------------- Captured log setup --------------------------------------------------------------
INFO     speculos-client:client.py:175 starting speculos with command: /home/lpascal/.virtualenvs/new/bin/python3 -m speculos --model nanos --sdk 2.1 --display headless /home/lpascal/repos/apps/boilerplate/app/bin/app.elf
```

It seems it triggers as soon as the version of Werkzeug is >= 2.1.0.

Will continue investigating.

greenknot, 2022-05-03:
Thanks a lot for the detailed analysis and for the patch!

lpascal-ledger, 2022-05-04:
Fixed and merged in [PR 283](https://github.com/LedgerHQ/speculos/pull/283)

closed 2022-05-04 by lpascal-ledger