LedgerHQ/speculos #304: How to connect ledger live

junjieit, 2022-06-21:


junjieit, 2022-06-21:
Sysyem version: Linux version 5.18.5-arch1-1 (linux@archlinux) (gcc (GCC) 12.1.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT_DYNAMIC Thu, 16 Jun 2022 20:40:45 +0000

Ledger live version: 2.43.1

Running speculos successfully with docker, but Ledger live doesn't recognize the device

greenknot, 2022-06-21:
Hello,

Short answer: it isn't possible to use speculos with Ledger Live (the [doc](https://speculos.ledger.com/user/clients.html#ledger-live-common) mentions a [tool](https://github.com/LedgerHQ/speculos/blob/8657b297798561a09d057b472f0e5d5fd9a58993/tools/ledger-live-http-proxy.py) but I don't think it works out-of-the-box.)

closed 2022-06-21 by greenknot