LedgerHQ/speculos #313: Emulator just displays a black screen

olalonde, 2022-07-16:
[image]

I tested with 2 apps, https://github.com/LedgerHQ/app-boilerplate and  https://github.com/LedgerHQ/app-recovery-check . I'm running speculos in docker on mac m1. There are no error messages anywhere (consoler or web app).

I tried building the speculos image both with `FROM ghcr.io/ledgerhq/speculos-builder-aarch64:latest AS builder` and 
`FROM ghcr.io/ledgerhq/speculos-builder:latest AS builder` but still same problem.

olalonde, 2022-07-16:
Example logs when I start speculos:

```
 * Serving Flask app 'speculos.api.api' (lazy loading) 
app-boilerplate-speculos-1  |  * Environment: development
app-boilerplate-speculos-1  |  * Debug mode: offed                                                                        0.1s
app-boilerplate-speculos-1  | 13:26:43.009:werkzeug:  * Running on all addresses.
app-boilerplate-speculos-1  |    WARNING: This is a development server. Do not use it in a production deployment.
app-boilerplate-speculos-1  | 13:26:43.010:werkzeug:  * Running on http://172.19.0.2:5000/ (Press CTRL+C to quit)
app-boilerplate-speculos-1  | [*] speculos launcher revision:
app-boilerplate-speculos-1  | [*] using SDK version 2.0 on nanos
app-boilerplate-speculos-1  | [*] loading CXLIB from "/speculos/speculos/cxlib/nanos-cx-2.0.elf"
app-boilerplate-speculos-1  | [*] patching svc instruction at 0x125c74
app-boilerplate-speculos-1  | [*] patching svc instruction at 0x40001c54
app-boilerplate-speculos-1  | [*] patching svc instruction at 0x40002668
app-boilerplate-speculos-1  | 13:26:44.137:seproxyhal.ticker: skipping ticker events (expected within a debugger)
```

After I go to http://localhost:5000:

```
 * Serving Flask app 'speculos.api.api' (lazy loading)
app-boilerplate-speculos-1  |  * Environment: development
app-boilerplate-speculos-1  |  * Debug mode: offed                                                                        0.1s
app-boilerplate-speculos-1  | 13:26:43.009:werkzeug:  * Running on all addresses.
app-boilerplate-speculos-1  |    WARNING: This is a development server. Do not use it in a production deployment.
app-boilerplate-speculos-1  | 13:26:43.010:werkzeug:  * Running on http://172.19.0.2:5000/ (Press CTRL+C to quit)
app-boilerplate-speculos-1  | [*] speculos launcher revision:
app-boilerplate-speculos-1  | [*] using SDK version 2.0 on nanos
app-boilerplate-speculos-1  | [*] loading CXLIB from "/speculos/speculos/cxlib/nanos-cx-2.0.elf"
app-boilerplate-speculos-1  | [*] patching svc instruction at 0x125c74
app-boilerplate-speculos-1  | [*] patching svc instruction at 0x40001c54
app-boilerplate-speculos-1  | [*] patching svc instruction at 0x40002668
app-boilerplate-speculos-1  | 13:26:44.137:seproxyhal.ticker: skipping ticker events (expected within a debugger)
app-boilerplate-speculos-1  | 13:29:01.153:werkzeug: 172.19.0.1 - - [16/Jul/2022 13:29:01] "GET / HTTP/1.1" 304 -
app-boilerplate-speculos-1  | 13:29:01.237:werkzeug: 172.19.0.1 - - [16/Jul/2022 13:29:01] "GET /screenshot HTTP/1.1" 200 -
app-boilerplate-speculos-1  | 13:29:01.317:werkzeug: 172.19.0.1 - - [16/Jul/2022 13:29:01] "GET /screenshot?force=1657978141304 HTTP/1.1" 200 -
app-boilerplate-speculos-1  | 13:29:01.318:werkzeug: 172.19.0.1 - - [16/Jul/2022 13:29:01] "GET /events?stream=true HTTP/1.1" 200 -
```

olalonde, 2022-07-16:
Ohh it seems the web app just takes screenshots of a GUI running in the docker process? I tried `--display qt` but now getting :

```
 ModuleNotFoundError: No module named 'PyQt5'
```

olalonde, 2022-07-16:
I also tried with `--text` but no UI is displayed:

[image]

olalonde, 2022-07-17:
I also tried running speculos in a x86 Ubuntu VM running in UTM. But getting different errors:

[image]

olalonde, 2022-07-21:
I tried in an Linux ARM VM and now it is launching the emulator but it's just a black screen...

[image]

olalonde, 2022-07-25:
It works with the `apps/btc.elf`.. I wonder why that is? Is it an issue with my compilation step?

greenknot, 2022-07-26:
Probably.

Since it isn't a speculos issue, please follow the developers documentation on the [Developer Portal](https://developers.ledger.com/docs/nano-app/introduction/), or go on [Discord](https://discord.com/invite/ledger) to chat with developer support and the developer community.

closed 2022-07-26 by greenknot