LedgerHQ/speculos #362: Any plan to add the BN_MONT_CTX family of functions?

hhanh00, 2023-03-29:
I think they would help speed up some custom ECC functions.

aido, 2024-01-20:
I am having issue with this also.

Speculos fails to implement `cx_mont_init2()` and `cx_mont_init()` causing my app to crash during testing. I am sure that if my app had continued without crashing I would also find that Speculos does not implement `cx_mont_mul()` either.

The previous comment says these functions are needed to speed up some ECC functions but I am finding them needed for some Galois Field functionality also.  See LedgerHQ/nanos-secure-sdk#64.
As pointed out in that issue and also in this issue, these functions are useful in several areas of cryptography.

It is quite an inconvenience not being able to test with Speculos the apps that use these functions.

aido, 2024-01-22:
It looks like someone has done some work on this already:

https://github.com/LedgerHQ/cylib/blob/master/wrappers/bolos/cy_wrap_fp_bolos.c

Can some of this code be used for inspiration?