LedgerHQ/speculos #415: Build fails on M2 Pro

vikinatora, 2023-08-07:
Hello, I followed the instructions for Apple Silicon.
- Replaced ```FROM ghcr.io/ledgerhq/speculos-builder-aarch64:latest AS builder```
- Ran `docker build ./ -t speculos`

During the build process I encountered the following error:
```
0.692 [  9%] Building C object src/CMakeFiles/emu.dir/bolos/cx_aes.c.o
0.792 [ 10%] Building C object src/CMakeFiles/emu.dir/bolos/cx_aes_sdk2.c.o
0.846 [ 10%] Building C object src/CMakeFiles/emu.dir/bolos/cx_blake2b.c.o
0.891 [ 11%] Building C object src/CMakeFiles/emu.dir/bolos/cx_bls.c.o
0.899 In file included from /speculos/src/bolos/cx_bls.c:2:
0.899 /speculos/src/bolos/cx_bls.h:4:10: fatal error: blst.h: No such file or directory
0.899     4 | #include <blst.h>
0.899       |          ^~~~~~~~
0.899 compilation terminated.
0.900 make[2]: *** [src/CMakeFiles/emu.dir/build.make:121: src/CMakeFiles/emu.dir/bolos/cx_bls.c.o] Error 1
0.900 make[2]: Leaving directory '/speculos/build'
0.900 make[1]: Leaving directory '/speculos/build'
0.900 make[1]: *** [CMakeFiles/Makefile2:1076: src/CMakeFiles/emu.dir/all] Error 2
0.901 make: *** [Makefile:114: all] Error 2
0.901 make: Leaving directory '/speculos/build'
------
Dockerfile:13
--------------------
  11 |     
  12 |     RUN cmake -Bbuild -H. -DPRECOMPILED_DEPENDENCIES_DIR=/install -DWITH_VNC=1
  13 | >>> RUN make -C build
  14 |     
  15 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c make -C build" did not complete successfully: exit code: 2
```
I'm running Docker Desktop 4.21.1 (114176) and on latest commit(0fed1b8de0fbaa721f1843685385724ed6e0a4b5)

chabroA, 2023-09-08:
So [the doc on dev portal is outdated](https://developers.ledger.com/docs/speculos/user/macm1/) and should be updated I guess?

lpascal-ledger, 2024-05-02:
Indeed the portal doc was partially deprecated. Should be better now.

closed 2024-05-02 by lpascal-ledger