LedgerHQ/speculos #488: Docker build fails on MacOS

scolear, 2024-05-21:
Trying to follow the documentation here for M1 Mac users: 
https://speculos.ledger.com/user/macm1.html#how-to-build-the-docker-image

Yields the following error:
```
0.648 /speculos/src/bolos/cx_bls.h:4:10: fatal error: blst.h: No such file or directory
0.648     4 | #include <blst.h>
0.648       |          ^~~~~~~~
0.648 compilation terminated.
0.648 make[2]: Leaving directory '/speculos/build'
0.649 make[1]: Leaving directory '/speculos/build'
0.649 make[2]: *** [src/CMakeFiles/emu.dir/build.make:121: src/CMakeFiles/emu.dir/bolos/cx_bls.c.o] Error 1
0.649 make[1]: *** [CMakeFiles/Makefile2:1098: src/CMakeFiles/emu.dir/all] Error 2
0.649 make: *** [Makefile:114: all] Error 2
0.649 make: Leaving directory '/speculos/build'
------
Dockerfile:13
--------------------
  11 |     
  12 |     RUN cmake -Bbuild -H. -DPRECOMPILED_DEPENDENCIES_DIR=/install -DWITH_VNC=1
  13 | >>> RUN make -C build
  14 |     
  15 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c make -C build" did not complete successfully: exit code: 2
```

Help would be appreciated!

scolear, 2024-05-22:
Thanks! 

I have gotten further this time following your description.
I can bash into the box successfully. However, running

```bash
docker run --rm -it -v $(pwd)/apps:/speculos/apps --publish 41000:41000 --publish 5001:5001 speculos --model nanos --display headless --vnc-port 41000 --api-port 5001 apps/btc.elf
```

Yield the following crash report:
```bash
09:55:36.549:root: Parsing binary '/speculos/apps/nanos#btc#2.1#1c8db8da.elf'
09:55:36.550:root: Parsing binary '/speculos/apps/nanos#btc#2.1#1c8db8da.elf'
09:55:36.603:speculos: Disabling OCR.
 * Serving Flask app 'speculos.api.api'
 * Debug mode: off
09:55:36.611:werkzeug: WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on http://127.0.0.1:5001
 * Running on http://172.17.0.2:5001
09:55:36.611:werkzeug: Press CTRL+C to quit
[*] speculos launcher revision: 
[*] using SDK version 2.1 on nanos
[*] loading CXLIB from "/speculos/speculos/cxlib/nanos-cx-2.1.elf"
[*] patching svc instruction at 0x126358
[*] Seed initialized from environment
launcher: using default app name & version
[*] patching svc instruction at 0x4000aebc
[*] patching svc instruction at 0x4000aeca
09:55:36.742:speculos: An error occurred
Traceback (most recent call last):
  File "/speculos/speculos/main.py", line 523, in main
    screen_notifier.run()
  File "/speculos/speculos/mcu/headless.py", line 84, in run
    self.notifiers[fd].can_read(self)
  File "/speculos/speculos/mcu/seproxyhal.py", line 338, in can_read
    if self.use_bagl and screen.display.screen_update():
  File "/speculos/speculos/mcu/headless.py", line 43, in screen_update
    return self.gl.refresh()
  File "/speculos/speculos/mcu/bagl.py", line 76, in refresh
    return self.fb.update()
  File "/speculos/speculos/mcu/headless.py", line 57, in update
    self._redraw()
  File "/speculos/speculos/mcu/headless.py", line 65, in _redraw
    self.vnc.redraw(self.pixels, self.default_color)
  File "/speculos/speculos/mcu/vnc.py", line 60, in redraw
    buf[i + 0] = y & 0xff
IndexError: bytearray index out of range
09:55:36.742:speculos: Stopping Speculos
launcher: read from seph fd failed: fd closed
```

is there something that needs to be set up _outside_ the container for this to work? I don't seem to find anything in the docs. Thanks!

oleonardolima, 2024-06-03:
FWIW: I was researching this one, as we've been facing the same issue, and the failure started to happen on version 0.8.4 and further, the `IndexError: bytearray index out of range` error seems consistent on all versions from 0.8.4 up to 0.9.1.